espressif/esp-dl #7: Detect movement only

wdouglas1999, 2019-05-31:
Good day, I use a PIR sensor to detect movement , but i receive many false triggers. I would like to know if there is a way to use your code to be able to compare consecutive images to confirm movement. I do not require facial recognition. 
Is there any library I could use with documentation available?
Thank you.

XiaochaoGONG, 2019-10-17:
Hi sorry for late responding. Did you find the solution ? For motion detection, you can use a very simple method that compare two consecutive images to find the region.

wdouglas1999, 2019-10-17:
Hi, thank you for your reply.
No, I did not find a solution. Do you have any example I may use?
Thank you.

XiaochaoGONG, 2019-10-18:
Not now, but in recent days, we will give a simple example.

xgarb, 2020-01-14:
Could you keep looking for movement for a set time to see if it's not a false trigger? Something like this where the device is woken by a PIR input and then checks 3 seconds later to see if the input is still triggering. You could do it in a loop and exit if the PIR goes low. You need a sensor that doesn't keep the input high too long after movement is detected.
 ```
 unsigned long timer_millis = millis();
  while (millis() - timer_millis < 3000) { //Test for movement for 3 seconds after warning
    if (digitalRead(PIR_PIN)) { // If still movement...
      do_something();
    }
 }
```
(full example: https://robotzero.one/ttgo-security-camera-pir/ )

Or did you want to check for the presence of a face after movement is detected? There's code here that might help: https://robotzero.one/unlock-door-face-detection-sonoff/

yehangyang, 2021-09-02:
Hi @wdouglas1999,

Is your problem solved?

wdouglas1999, 2021-09-02:
Hi, I have not yet solved this issue due to time constraints. However I have seen an Arduino based project focusing on this type of movement detection using various methods. I was looking for something I could use in IDF environment though.

yehangyang, 2021-09-03:
Hi @wdouglas1999,

Please check this API [`get_moving_point_number()`](https://github.com/espressif/esp-dl/blob/8a5be0973686e8ae16e19a345025a954b5cdc31e/include/image/dl_image.hpp#L326) in ESP-DL.

I hope this can meet your requirement.

wdouglas1999, 2021-09-03:
Very interesting, I will look into it for sure. Thank you.

closed 2021-09-03 by wdouglas1999

yehangyang, 2021-09-04:
Hi @wdouglas1999 ，

We'll provide an example in ESP-WHO later.