espressif/esp-dl #9: Issue while reading file from SD card

Kirankumar-xenlabs, 2019-10-11:
Hello, 
I want to read a simple text file, after face detection, system crashed when fread function called, writing a txt file/ image file it is working well, I pasted the backtrace please refer

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d8404  PS      : 0x00060230  A0      : 0x800d84fc  A1      : 0x3ffc4da0
0x400d8404: _fread_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fread.c:194

A2      : 0x3ffc4f6c  A3      : 0x3ffc4e00  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3ffb22a8  A8      : 0x00000001  A9      : 0x3ffc4d90
A10     : 0x3ffc4f04  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x000001b6
A14     : 0x00000072  A15     : 0x0000002b  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000064  LBEG    : 0x40090ac0  LEND    : 0x40090acb  LCOUNT  : 0x00000000
0x40090ac0: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x40090acb: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x400d8404:0x3ffc4da0 0x400d84f9:0x3ffc4dd0 0x400d59d9:0x3ffc4df0 0x400891a9:0x3ffc4e70
0x400d8404: _fread_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fread.c:194

0x400d84f9: fread at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fread.c:301

0x400d59d9: task_process at C:\Users\XL008\Desktop\esp-idf\esp-who-master\examples\single_chip\fd_sd\build/../main/app_facenet.c:156

0x400891a9: vPortTaskWrapper at C:/Users/XL008/Desktop/ESP/esp-idf-2/components/freertos/port.c:435

Thanks & Regards,
Kiran

igrr, 2019-10-11:
Looks like a NULL pointer is passed to fread. Suggest enabling GDB Stub and checking the arguments in backtrace.

Kirankumar-xenlabs, 2019-10-11:
@igrr thank you for reply, how do I enable that, by using menuconfig?

XiaochaoGONG, 2019-10-17:
Or you can debug with 'printf' :)

closed 2019-11-20 by Kirankumar-xenlabs