espressif/esp-dl #11: While linking undefined reference to functions from dl_lib_matrix3d.h and dl_lib_matrix3dq.h

rucebee, 2019-11-04:
For example:
<HOME>/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /<MY_ESP_HOME>/esp-who/components/esp-face/lib/libod.a(sparse_mn_5_q.c.obj):(.literal.sparse_mn_5_q_without_landmark+0x40c): undefined reference to `dl_matrix3dqq_mobilenet'

Could you please suggest a solution?

XiaochaoGONG, 2019-11-05:
In what situation did you meet that ? Please add more information, or is there a problem with the default esp-who example ?

rucebee, 2019-11-05:
Thank you for prompt reply
Yes, it's a problem with the default esp-who example (Should I open this issue at esp-who git repo?).
When ESP-WHO Face Detection is enable in Camera Web Server config, it fails to build at linkage (otherwise it's ok).
I guess libdl_lib.a misses something.
BTW, I use esp-idf previously installed to my esp root dir (not the one attached to esp-who git) but without face detection all builds well

XiaochaoGONG, 2019-11-05:
Hi, thanks for finding the bug, please update the esp-who with 27e8cde. It may solve your problem.

rucebee, 2019-11-05:
[steps.txt](https://github.com/espressif/esp-face/files/3809275/steps.txt)
Did not helped, see my steps.txt attached

Did I make git update wrong?

XiaochaoGONG, 2019-11-06:
call `git submodule update`

rucebee, 2019-11-06:
thanks you for being supportive

the issue was that I used the latest extensa toolchain (which is installed with latest esp-idf)
so to build successfully it is not enough to switch to esp-who/esp-idf
but one also needs to switch to previous toolchain

I succeeded with https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz

closed 2019-11-06 by rucebee