espressif/esp-dl #12: Cannot invoke 3X3 convolution in Arduino IDE

caleb221, 2019-11-05:
Hi!
I am trying to invode dl_matrix3dqq_conv_3x3() in the Arduino IDE in order to do some deep learning classification but the compiler cant find the function in the library, what do i do??
after creating a quantized matrix I can use other functions in the library (ReLu, PreRelu, allocation, etc..) but when it comes to the convolution operations I get the following error:

ESP32_DeepNet_attempt:219:12: error: 'dl_matrix3dqq_conv_3x3' was not declared in this scope
outMatrix=dl_matrix3dqq_conv_3x3(qMatrix,qMatrix,2,2,0,2,"test");
^
exit status 1
'dl_matrix3dqq_conv_3x3' was not declared in this scope

when it is defined in dl_lib_matrix3dq.h library as:

/**

@brief Do 1x1 convolution with an 8-bit fixed point matrix, with bias adding
@param out Preallocated quantized matrix, size (1, w, h, n)
@param in Input matrix, size (1, w, h, c)
@param filter 1x1 filter, size (n, 1, 1, c)
@param bias Bias, size (1, 1, 1, n)
@param mode Implementation mode
*/
void dl_matrix3duq_conv_1x1_with_bias(dl_matrix3dq_t *out,
dl_matrix3du_t *in,
dl_matrix3dq_t *filter,
dl_matrix3dq_t *bias,
dl_conv_mode mode,
char *name);
//
// Conv 3x3
//

Hardware:
Board: ESP32-Cam
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 40Mhz
PSRAM enabled: yes
Upload Speed: 115200
Computer OS: Ubuntu

XiaochaoGONG, 2019-11-05:
Hi, since you can refer to other functions declared in the `dl_matrix3dq.h`, you should normally can use `dl_matrix3dqq_conv_3x3` as well. Can you use other `1x1` or `3x3` functions ?

caleb221, 2019-11-06:
Every function defined in the .h as
dl_matrix3dq_t *dl_matrix...... cannot be used,
but any function defined with 
void dl_matrix...  can be used
I'm not sure what I'm doing wrong

XiaochaoGONG, 2019-11-06:
Seems the compiler doesn't know the dl_matrix3dq_t * functions... What is the compiler the Arduino IDE using ? And one thing need to be confirmed that is make or idf.py(cmake) also meet that strange thing? I've tested on my side without any problem yet.

caleb221, 2019-11-07:
yeah, the compiler should be using the same as espressif V1.0.0.4 (gcc i believe)
 using the program
void setup(){
  Serial.begin(115200);
  Serial.print("C++ STANDARD: ");
  Serial.println(__cplusplus);
}
void loop(){}

 outputs: C++ VERSION: 201103

Another odd thing, when i tried directly moving the library into the project folder I got errors concerning redefinitions of the same dl_matrix_3dq_t* functions, so the compiler should know the functions exist (it does when there are 2 of them) but it won't let me invoke them. 

Thanks for your help on this, its really odd and I need to get a model up and running for a research project thats due in a few weeks

XiaochaoGONG, 2019-11-08:
Try adding `extern "C"` before the declaration of those funcions in header files. The libraries are build with C, the C++ compiler would adding some suffix to distinguish their override functions.

closed 2019-11-10 by caleb221