espressif/esp-dl #13: esp-face -> esp-object (DL - Component)

andybru, 2019-11-05:
Hello and thank you for the very interesting features you provide
with the esp-face component.

Hopefully probably soon you provide more insight into the DL(DeepLearning)-Framework, that you are using.

Will there be more details how to train other objects to be detected?

It would be very comforting if I possibly could train an esp-camera
for any or say some more specialized objects like balls, pyramids, cars or teddy bears.

Especially if these detections could be made a little faster for more simple objects
the field of application(i.e. Robot-Arm-Grasping) of my esp-camera would rise enormously.

Would you mind to give any time-frame when you explain in more detail your
DL-Framework?

Thank you very much so far.

XiaochaoGONG, 2019-11-20:
Thanks for your appreciation. There will be some descriptions in detail about how to implement users' own model and how to use it.

andybru, 2019-11-21:
Thank you for quick comment!

I know it is somehow a very hot topic in the forum
and others asked similar questions.

And to have all this already functioning out of the box is really cool!!

by caleb221, on Tue Nov 05, 2019 8:34 am, suggested the following
(  https://www.esp32.com/viewtopic.php?t=12831#p51388  ) ..

Is he quite right?

> .. there will be some descriptions
You wrote this some time ago in other comments too. 
You also mentioned somewhere a pending patent ..

Would you mind to mention here some probable time-frame for better documentation?

something like i.e   Descriptions possibly in six months?

Or a year or so?

Anyway good luck and greets from Dresden

XiaochaoGONG, 2020-01-14:
Hi please update to 8a6e07ae102aeca1f3ee8d418804d8914952d4c9, a simple tutorial to implement customized model.

closed 2021-09-02 by TiramisuJ