espressif/esp-dl #15: assertion "out->c == bias->c" failed:

caleb221, 2019-11-27:
assertion "out->c == bias->c" failed: file "/home/gongxiaochao/code/esp/esp_temp/dl_lib/components/dl_lib/dl_lib_matrix3d.c", line 37, function: dl_matrix3d_init_bias
abort() was called at PC 0x400e1c43 on core 1

XiaochaoGONG, 2019-11-28:
Hi Caleb, while you're doing convolution with bias adding, you need to ensure that the channel of output and the channel of bias is same. So what is the dimension of each ?

caleb221, 2019-11-28:
Ok, so calling the init_bias function with these parameters works out fine but inside the convolution function it fails  

 dl_matrix3d_t *filt_c1 = dl_matrix3d_alloc(10,1,1,1);
  dl_matrix3d_t *bias1 = dl_matrix3d_alloc(10,1,1,1);

  dl_matrix3d_t *net_out= dl_matrix3d_alloc(10,1,1,1);
  dl_matrix3d_t *out1;

  filt_c1 = getConv_filter1(); //filter values from a .h file i created
  bias1 = getBias1();
  dl_matrix3d_init_bias(net_out,bias1);//THIS WORKS (?) 
    int width = round(img->w * 12);
    int height = round(img->h * 12);

    dl_matrix3du_t *in = dl_matrix3du_alloc(1, width, height, img->c);
   printf("Resizing...\n");
   image_resize_linear(in->item, img->item, width, height, in->c, img->w, img->h);

        // 3x3 convolution #1
        //convert 2 3d
        dl_matrix3d_t *try = transformImage(in); // change the type from dl_matrix3du_t to dl_matrix3d_t
        dl_matrix3du_free(img); // don't need the original data, its eating up memory
         // THIS FAILS ='(
        out1= dl_matrix3dff_conv_3x3(try,filt_c1,bias1,10,10,PADDING_SAME);

caleb221, 2019-11-28:
I realize trying to run with (10,1,1,1) bias and filter passes the init_bias function, and fails with the convolution function, but when i change all the dimensions to (10,1,1,3) (in order to match the image) i cannot initialize the bias

caleb221, 2019-11-28:
I think it might be a misundertanding in how to properly format my data.

[image]

-

caleb221, 2019-11-28:
[image]
I think it might be in my understanding of the data structure.
I have tried to draw it out in the picture above, if it is incorrect please let me know
in order to translate a matrix from python (numpy) that has shape (10,3,3,3) to the dl_matrix library I do the following
reshape the array to 1 dimension and copy the values (float32) into a header file.
in that header file i decare a dl_matrix3d_t of shape (1,3,3,3)
then set the stride to 10 (because of the original numpy shape)
and set all items in the matrix as the values from the numpy array
return the dl_matrix

the bias is done using the same process above
--> the number of values is very different though (items is much smaller)

is this correct?

XiaochaoGONG, 2019-11-29:
Ah I see, in term of output for convolution, the N demiension is always 1, i.e. output: (1, w', h', c'), where c' = # of filters. That also fit to bias, i.e. bias: (1, 1, 1, c'), where c' = # of filters.

XiaochaoGONG, 2019-11-29:
And `dl_matrix3d_init_bias` is to put bias value into the output to improve the performance.

caleb221, 2019-11-29:
that worked thank you so much!

closed 2019-11-29 by caleb221

XiaochaoGONG, 2019-11-29:
Be careful that the weight data in our system is in NHWC format while in numpy is usually HWCN format. And the stride value here is W\*C. When you call `dl_matrix3d_alloc`, the stride will be automatically set.

caleb221, 2019-11-29:
Oh! I hadn't realized that, Thanks!
So, instead of simply 
translated=numpy.reshape(weight_array,-1)
it would be
translated=numpy.moveaxis(weight_array,1,3)
copy_out=numpy.reshape(translated,-1)
and copy_out goes to the header file

caleb221, 2019-11-30:
oops, taking a second look i was wrong with initial translations..
should be 
            t=np.moveaxis(c,3,0) # translate from (H,W,C,N) caffe/numpy to (N,H,W,C) EPS lib         
            # out: ( n,w,c,h ) 
            #next -> c,h (2,3)
            t1 = np.moveaxis(t,2,3)
            #out: (n,w,h,c)
            #next --> 1,2
            t = np.moveaxis(t1,1,2)
            # out --> n,h,w,c woohoo! translated
I think...