espressif/esp-dl #18: Matrix Slicing

caleb221, 2019-12-11:
Hello!
I would like to make sure I am using the dl_matrix3d_slice_copy() function correctly.

Suppose I have a matrix with the shape (1,1,1,48)
And I would like to slice the second half into a copy.
Proper usage would look like this:
input shape = (1,1,1,48)
output shape = (1,1,1,24)

dl_matrix3d_slice_copy(output,input,24,2,1,1);

This code does execute, but i believe it is poisoning my heap as I will get an error later.
I have tried multiple combinations of numbers 
but one thing remains unclear.
is the x-offset the end point and w the start point?
Thanks!

XiaochaoGONG, 2019-12-12:
This slice only works on planar, i.e. slicing in width and height dimension, all channels remain.

caleb221, 2019-12-14:
Oh I think i see,
so if i wanted half of a 3dmatrix (same channels) it would be something like
input shape: (1,12,12,28)

        dl_matrix3d_t *half_1 = dl_matrix3d_alloc(1,6,6,14);
        dl_matrix3d_t *half_2 = dl_matrix3d_alloc(1,6,6,14);
        int x = (input->h)/2;
        int y = (input->w)/2;
        dl_matrix3du_slice_copy(half_1, input, x, y, 6, 6);
        dl_matrix3du_slice_copy(half_2, input, x*2, y*2, 6, 6);

XiaochaoGONG, 2019-12-17:
> dl_matrix3du_slice_copy(half_1, input, x, y, 6, 6);
> dl_matrix3du_slice_copy(half_2, input, x*2, y*2, 6, 6);

```
dl_matrix3du_slice_copy(half_1, input, 0, 0, 6, 6);
dl_matrix3du_slice_copy(half_2, input, x, y, 6, 6);
```

caleb221, 2019-12-17:
oh ok awesome! Thank you!

closed 2019-12-17 by caleb221