espressif/esp-dl #23: How to store enrolled faces on Firebase instead of on flash?

trelelelel02, 2020-01-20:
I'm currently doing facial recognition for my office, and given the number of people, I thought it would be better to store the data of the enrolled faces and their matching ids in a firebase database. However even  after reading all the documentation, I still don't know what specific values of variables I need to send to the database, as well as what I would call from the database.

XiaochaoGONG, 2020-02-07:
Hi, when you called `enroll_face()`, you will get a list of face id if the confirm time is satisfied.
https://github.com/espressif/esp-face/blob/9fc6796326494cbe883eb786d86a0bc36eb7950b/face_recognition/fr_forward.c#L352-L385

The `face_id_list` is a struct that contains the face ids.
https://github.com/espressif/esp-face/blob/9fc6796326494cbe883eb786d86a0bc36eb7950b/face_recognition/include/fr_forward.h#L51-L59

So back to your question, you can access face ids via `face_id_list.id_list[n]`, e.g. go through the entire list:
```c
face_id_list *l = &face_id;
for (int i = 0; i < l->count; i++)
{
    uint8_t head = (l->head + i) % l->size;
    dl_matrix3d_t *face_id_vector = l->id_list[head];
    ...    // you can process the face_id_vector, its dimension is (1, 1, 1, 512)
}
```

eesati, 2020-12-06:
> I'm currently doing facial recognition for my office, and given the number of people, I thought it would be better to store the data of the enrolled faces and their matching ids in a firebase [...]

Hello  @trelelelel02 did you manage to do that, how far have you proceeded with that ? 

Thank you

yehangyang, 2021-09-06:
Hi @trelelelel02 ,

Is this problem solved?

closed 2021-09-28 by TiramisuJ