espressif/esp-dl #24: when build a project include esp-face components ,there is many build error like this?

zhujiapin2018, 2020-02-05:
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:835: undefined reference to `dl_matrix3dqq_depthwise_conv_3x3s1_with_bias'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:842: undefined reference to `dl_matrix3dqq_mobilefaceblock'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:846: undefined reference to `dl_matrix3dqq_mobilefaceblock'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:865: undefined reference to `dl_matrix3dqq_mobilefaceblock'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:884: undefined reference to `dl_matrix3dqq_mobilefaceblock_split'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:903: undefined reference to `dl_matrix3dqq_mobilefaceblock_split'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:924: undefined reference to `dl_matrix3dqq_mobilefaceblock_split'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:945: undefined reference to `dl_matrix3dqq_mobilefaceblock_split'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:966: undefined reference to `dl_matrix3dqq_mobilefaceblock_split'
c:/progra~1/espres~1/esp-id~1/tools/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/Administrator/esp/m5stack-cam-psram-master/mpu6050/components/esp-face/lib\libfr.a(frmn.c.obj):/home/gongxiaochao/code/esp/esp_temp/dl_lib/components/face_recognition/frmn/frmn.c:987: more undefined references to `dl_matrix3dqq_mobilefaceblock_split' follow

XiaochaoGONG, 2020-02-08:
Hi can you provide more details and your operations on the building. It might help you if you do `idf.py fullclean` to clean the old build.

zhujiapin2018, 2020-02-09:
非常感谢您的回复，可能是我搭建的idf环境有问题，我重新搭建可一个环境，没有问题了，非常感谢！发自我的华为手机

closed 2020-03-31 by XiaochaoGONG