espressif/esp-dl #26: In <dl_matrix3dff_conv_common> function the bias weights are not taken into account.

Vvard, 2020-03-11:
[image]
[image]
[image]
[image]
[image]
[image]

I used "dl_matrix3dff_conv_common(dl_matrix3d_t *in, dl_matrix3d_t *filter, dl_matrix3d_t *bias, int stride_x, int stride_y, dl_padding_type padding)" function with different bias weights, but there are no changing results. This is a small test program with 4x4 input data, where I run the program in the case of 0 bias weight and in the case of 1 bias weight and I get the same result.

XiaochaoGONG, 2020-03-11:
Good test ! I'll fix that.

XiaochaoGONG, 2020-03-12:
f1a3dbe8ffe418aa6f9372831f9c1e33a651ab5b Please update and try.

Vvard, 2020-03-12:
Works correctly! thanks)

closed 2020-03-12 by Vvard