espressif/esp-dl #27: PADDING_SAME_DONT_FREE_INPUT padding type description

Vvard, 2020-03-17:
Could you describe the padding type PADDING_SAME_DONT_FREE_INPUT in dl_lib matrix3d.h?

Vvard, 2020-03-24:
And there is complete documentation for the DL API?

XiaochaoGONG, 2020-03-24:
Hi I'll update the documentation of dl_lib. Please wait for a while, maybe next week.
For short, the PADDING_SAME flag indicates the process would free the input matrix when finish padding, while PADDING_SAME_DONT_FREE_INPUT indicates the input matrix remain. Both of PADDING_SAME will give you a new allocated padded input.
- PADDING_SAME: input --> return malloc padded input and free input
- PADDING_SAME_DONT_FREE: input --> return malloc padded input

Vvard, 2020-03-24:
OK, thanks)

closed 2020-03-24 by Vvard