espressif/esp-dl #29: Where is the (.c)file in arduino IDE?

Rita-GE, 2020-07-22:
board : ESP32-CAM
environment ：arduino IDE(1.8.10)
esp32 core : 1.0.4
PC：win10

When I want to learn the function of face detection and recognition yesterday，I found that there was no (.c) file to explain this function.
Finally,I found [here](https://github.com/espressif/esp-face).

However,what make me confuse is how can arduino IDE manage to compile without (.c) function file.
Or where do I find its (.c)file.(More important)

More detail:
The path of face detection and recognition lib in my computer:
`C:\Users\ASUS\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp-face`

The contents of the folder are shown below:
[image]

It's really amazing. 
It would be a great help if someone could help.
Thank you!

TiramisuJ, 2020-07-22:
Hi, we have compiled these (.c) files into (.a) files. You can find them in the 'lib' folder under the 'esp-face'

Rita-GE, 2020-07-22:
> Hi, we have compiled these (.c) files into (.a) files. You can find them in the 'lib' folder under the 'esp-face'

I see.
Thank you for your help.

maartenvds, 2021-02-24:
@TiramisuJ Why don't you provide the .c files? Is it for proprietary reasons or for convenience? It would be nice if the source files would be available for the community. I'm for instance interested in the convolution kernels, how they are written. Do you use any kind of MAC or DSP like instructions to accelerate the computation?

closed 2021-09-02 by TiramisuJ