espressif/esp-dl #30: About face detection and face recognition

Rita-GE, 2020-08-03:
Hi,
I just wonder: 
1. Where is the "face id vectors" exactly store? (flash or RAM?)
2. And how many face id can it store?
3. Does face recognition use a one-to-one comparison to get the owner of the face or a one-to-many method to find the person?
4. What is the value of FR and FA?
5. How long does it take on face detection? What about face recognition?

TiramisuJ, 2020-08-03:
Hi
1. In CameraWebServer, The face id vectors are stored in RAM.
2. You can set the maximum number of face IDs that can be stored, in the CameraWebServer, you can set the 'FACE_ID_SAVE_NUMBER' in 'app_httpd.cpp'.
3. The face recognition will compare the face information that has been enrolled so far.
4. Sorry i didn't understand this question. FD and FR mean face detection and face recognition respectively.
5. You can refer to [face_detection](https://github.com/espressif/esp-face/blob/master/face_detection/README.md) and [face_recognition](https://github.com/espressif/esp-face/blob/master/face_recognition/README.md) for more details.

Rita-GE, 2020-08-04:
> Hi
>
> 1. In CameraWebServer, The face id vectors are stored in RAM.
> 2. You can set the maximum number of face IDs that can be stored, in the CameraWebServer, you can set the 'FACE_ID_SAVE_NUMBER' in [...]

Great, thank you.
By the way, FR means false acceptance rate and FA means false rejection rate. 
Can you tell me their values?

TiramisuJ, 2020-08-04:
1. FAR and FRR are not independent of each other, and the relationship between them can be represented by the ROC curve. For different test sets, the ROC curve of our model is also different. The FRR corresponding to a fixed FAR under a specific test set may be a more reasonable performance indicator.
2. ‘EYE_DIST_SET' is a parameter that we used for face alignment before, and it is used to indicate the distance between the two eyes of the aligned face. But we have adopted a new face alignment method, and no longer use the ‘EYE_DIST_SET’ parameter.

Rita-GE, 2020-08-04:
> 1. FAR and FRR are not independent of each other, and the relationship between them can be represented by the ROC curve. For different test sets, the ROC curve of our model is also different. The [...]

Then, does “FACE_ID_SIZE” mean the size of "The face id vectors"?

TiramisuJ, 2020-08-04:
> > 1. FAR and FRR are not independent of each other, and the relationship between them can be represented by the ROC curve. For different test sets, the ROC curve of our model is also different. The FRR corresponding to a fixed FAR under a specific test set may be a more reasonable performance indicator.
> > 2. ‘EYE_DIST_SET' is a parameter that we used for face alignment before, and it is used to indicate the distance between the two eyes of the aligned face. But we have adopted a new face alignment method, and no longer use the ‘EYE_DIST_SET’ parameter.
> 
> Then, could you tell me that does “FACE_ID_SIZE” means the size of "The face id vectors"?
> Or what is the fixed max size of "The face id vectors"?

‘FACE_ID_SIZE’ is the length of each face vector. This value depends on the face recognition model. At present, the length of the face vector obtained by our model is 512.

Rita-GE, 2020-08-04:
> ‘FACE_ID_SIZE’ is the length of each face vector. This value depends on the face recognition model. At present, the length of the face vector obtained by our model is 512.

How many faces can it detect in a frame?

Rita-GE, 2020-08-12:
Hi, it's been a long time.
I'm here to bother you again.

When I test the size of the new face ID, there is a strange result below:
[image]
I remember that you said that the size is 512.

This is my code below:
```
Serial.printf("Enrolled Face ID: %d\n", id_list.tail);
Serial.printf("Enrolled Face ID Size: %d\n", sizeof(id_list.id_list[id_list.tail]));
Serial.printf("Enrolled Face ID list Size: %d\n", sizeof(id_list));
```
Did I do it wrong?
And how can I get it?

Hope to receive your answer soon.
Thank you!

TiramisuJ, 2020-08-12:
1. ‘id_list.id_list[id_list.tail])’ is a pointer to a 'dl_matrix3d_t' structure，so the size of the pointer is 4, and the face vector with a length of 512 is stored in the 'dl_matrix3d_t' structure.
2. 'id_list' is a 'face_id_list' structure, the size of the 'dl_matrix3d_t **' is 4, So this structure is 4 bytes aligned, and the size of 'face_id_list' structure is 12.

Rita-GE, 2020-08-13:
So nice you are.Thank you.
Finally, I got that size.

Rita-GE, 2020-08-13:
> 1. ‘id_list.id_list[id_list.tail])’ is a pointer to a 'dl_matrix3d_t' structure，so the size of the pointer is 4, and the face vector with a length of 512 is stored in the 'dl_matrix3d_t' structure. [...]

So nice you are.Thank you.
Finally, I got that size.

kr817, 2020-12-04:
> > 1. “ id_list.id_list [id_list.tail]）”是指向“ dl_matrix3d_t”结构的指针，因此该指针的大小为4，长度为512的面部向量存储在“ dl_matrix3d_t”结构中。
> > 2. 'id_list'是'face_id_list'结构，'dl_matrix3d_t **'的大小为4，因此此结构是4个字节对齐的，而'face_id_list'结构的大小是12。
> 
> 太好了，谢谢。
> 终于，我得到了那个尺寸。

> So nice you are.Thank you.
> Finally, I got that size.

> So nice you are.Thank you.
> Finally, I got that size.

How did you get that size？

kr817, 2020-12-04:
> So nice you are.Thank you.
> Finally, I got that size.

How did you get that size

closed 2021-09-02 by TiramisuJ