espressif/esp-dl #38: Precompiled detection .a files in Arduino IDE

robotzero1, 2021-01-14:
Should I be able to use the libdetection_cat_face.a file in the Arduino IDE? Maybe using it as a library with the precompiled=true property?

I get this on compilation

```
Compiling library "cat_face_3"
Library cat_face_3 has been declared precompiled:
Using precompiled library in C:\Users\Dude\Documents\Arduino\libraries\cat_face_3\src\esp32
The plaform does not support 'compiler.libraries.ldflags' for precompiled libraries.

```

The ESP32 crashes at this point:
`assertion "model->model_config.enabled_top_k > 0" failed`

So I guess the model isn't loading but I'm not sure.

I have the cat face detection working using the IDF but I would like to make a little project with it using the Arduino IDE to make it easier for people to play with.

yehangyang, 2021-09-02:
Hi @robotzero1,

Sorry for the late reply.
Is the issue solved?

robotzero1, 2021-09-02:
Not really. I got as far as this and moved on to something else because I wasn't sure if was even possible.

me-no-dev, 2021-09-02:
we have thought of this, but have not found a way to implement it in Arduino IDE. BTW this issue should have been filed in the ESP32 Arduino repo :)

closed 2021-09-02 by me-no-dev