espressif/esp-dl #40: Bug: dl_matrix3dff_conv_3x3 gives incorrect output, when n > 1.

Vvard, 2021-03-26:
I compared the dl_matrix3dff_conv_3x3 and dl_matrix3dff_conv_common functions, but the dl_matrix3dff_conv_3x3 function gives incorrect output when n> 1.

[conv_test.txt](https://github.com/espressif/esp-face/files/6211245/conv_test.txt)
[image]

TiramisuJ, 2021-03-29:
Could you show me your code? I can't get your results in my project.

Vvard, 2021-03-29:
Hello @TiramisuJ,
My code is in conv_test.txt file.

[conv_test.txt](https://github.com/espressif/esp-face/files/6221917/conv_test.txt)

#include <stdio.h>
#include <stdlib.h>
#include "dl_lib_matrix3d.h"

#define HEIGHT 4
#define WIDHT 4

const static fptp_t Conv3D_bias_item_array[] = {
	0.0f, 0.0f
};

const static dl_matrix3d_t Conv3D_bias = {
	.w = 1,
	.h = 1,
	.c = 2,
	.n = 1,
	.stride = 2,
	.item = (fptp_t *)(&Conv3D_bias_item_array[0])
};

const static fptp_t Conv3D_kernel_item_array[] = {
	0.0f, 0.0f, 0.0f,
	0.0f, 1.0f, 0.0f,
	0.0f, 0.0f, 0.0f,
	0.0f, 0.0f, 0.0f,
	0.0f, 0.0f, 0.0f,
	0.0f, 0.0f, 0.0f
};

const static dl_matrix3d_t Conv3D_kernel = {
	.w = 3,
	.h = 3,
	.c = 1,
	.n = 2,
	.stride = 6,
	.item = (fptp_t *)(&Conv3D_kernel_item_array[0])
};

void printData(dl_matrix3d_t);

void app_main()
{
	dl_matrix3d_t *input = dl_matrix3d_alloc(1, HEIGHT, WIDHT, 1);

	float inputData[] = { 1.0f, 1.0f, 1.0f, 0.0f, \
						  0.0f, 1.0f, 1.0f, 1.0f, \
						  0.0f, 0.0f, 1.0f, 1.0f, \
						  0.0f, 0.0f, 1.0f, 1.0f
	};

	input->item = inputData;

	puts("Input Data");
	printData(*input);

	puts("Conv3d Kernel");
	printData(Conv3D_kernel);

	puts("Conv3d Bias");
	printData(Conv3D_bias);

	dl_matrix3d_t *Conv3D_1 = dl_matrix3dff_conv_common(input, &Conv3D_kernel, &Conv3D_bias, 1, 1, PADDING_VALID);

	dl_matrix3d_t *Conv3D_2 = dl_matrix3dff_conv_3x3(input, &Conv3D_kernel, &Conv3D_bias, 1, 1, PADDING_VALID);

	puts("dl_matrix3dff_conv_common");
	printData(*Conv3D_1);

	puts("dl_matrix3dff_conv_3x3");
	printData(*Conv3D_2);
}

void printData(dl_matrix3d_t data)
{
	int n = data.n;
	int w = data.w;
	int h = data.h;
	int c = data.c;

	for (int b = 0; b < n; ++b) {
		printf("n = %d\n", b);
		for (int k = 0; k < c; ++k) {
			printf("c = %d\n", k);
			for (int j = 0; j < w; ++j) {
				for (int i  =0; i < h; ++i)
					printf("%.1f\t", data.item[b * w * h * c + j * h * c + i * c + k]);
				puts(" ");
			}
		}
		puts(" ");
	}
	puts(" ");
}

Regards,
Vvard

TiramisuJ, 2021-03-30:
Hi, The "stride" parameter in "dl_matrix3d_t" structure is equal to w * c. So the initial value of the "stride" in "Conv3D_kernel" is wrong, and the correct value is 3. You can try again after fixing this bug.

TiramisuJ, 2021-03-30:
Another tip. When you use the "dl_matrix3d_alloc" function, it will malloc a space for the "item". Then you change the address of the "item" pointer. If you want to free your "input", don't forget to free the item space that "dl_matrix3d_alloc" function applies.

Vvard, 2021-03-31:
Hi @TiramisuJ,

Thanks a lot for your assistance.
That helped.

Actually I ported a code implemented by dl_matrix3dff_conv_common, which seems does not require correct assignment for stride parameter, which mislead and caused the incorrect understanding of "stride" parameter.
This might not be crucial, but you might consider fixing it in future.

Thanks again.
Have a nice day)

TiramisuJ, 2021-03-31:
Hi @Vvard ,
The implementation of the "dl_matrix3dff_conv_common" function and the "dl_matrix3dff_conv_3x3" function is different. For a 3x3 convolution calculation, using "dl_matrix3dff_conv_3x3" is faster. 
Thanks for your advice. I will fix it in future.

Vvard, 2021-04-01:
Hi @TiramisuJ 

Thank you for advice)

Regards,
Vard

closed 2021-04-01 by Vvard