espressif/esp-dl #41: Please Help!!!! - mfn56 code - feature extractions

AD121390, 2021-04-29:
I am having trouble finding the code for mfn56. I am looking for the code because I need to know how many feature extractions it does, and what are the features extracted? I know face_id is a vector of length 512. I do not think all of them are geometric facial features, perhaps skin tone/color? I am using esp-face on my esp32 for a school project, and my professor wants my group to do the facial recognition part.  I know that face_id has the feature extractions that I need for facial recognition. I just do not know what those 512 elements are measuring. 

Any help is appreciated. Thank you.

TiramisuJ, 2021-05-06:
Hi, This 512-dimensional vector is the high-dimensional feature of the face. we did not constrain the specific face information represented by each dimension. But in a high-dimensional space, the distance between two vectors of the same person is very small, while the distance between two vectors of different people will be very large.

AD121390, 2021-05-06:
I am having a problem where the face_id vector is the same no matter what face is detected by the esp32 camera. How can I fix this???

TiramisuJ, 2021-05-06:
First of all, make sure that the image obtained by the camera is correct, and then use the API we provide to extract the features of the face. You can refer to this [https://github.com/espressif/esp-who/tree/master/examples/single_chip/face_recognition_with_command_line](demo) to write your code