espressif/esp-dl #44: Matrix multiplication

PureHing, 2021-08-11:
@yehangyang
Hi，How to use [mul2d](https://github.com/espressif/esp-dl/blob/v2_alpha/include/nn/dl_nn_mul2d.hpp) to multiply two matrices?

```bash
    __attribute__((aligned(16))) int16_t element1[] = {5, 6, 7, 8};
    __attribute__((aligned(16))) int16_t element2[] = {1, 2, 3, 4};
    Tensor<int16_t> input1;
    input1.set_element((int16_t *)element1).set_exponent(0).set_shape({2,2,1}).set_auto_free(false);
    Tensor<int16_t> input2;
    input2.set_element((int16_t *)element2).set_exponent(0).set_shape({2, 2, 1}).set_auto_free(false);
    Tensor<int16_t> output;
    dl::nn::mul2d(output,input1,input2);
```
Is this reasonable?

yehangyang, 2021-08-11:
Hi @PureHing 

In your code, it actually calls this [void mul2d(Tensor<int16_t> &output, Tensor<int16_t> &input0, Tensor<int16_t> &input1, const Activation<int16_t> *const activation = NULL, const std::vector<int> &assign_core = CONFIG_DEFAULT_ASSIGN_CORE);](https://github.com/espressif/esp-dl/blob/cc615c23793efffb5380b32b2c3d3eae54df7e98/include/nn/dl_nn_mul2d.hpp#L20). In this way, we have make sure the output has been initialized with shape, exponent, etc. Take [dl_layer_mul2d.hpp](https://github.com/espressif/esp-dl/blob/v2_alpha/include/layer/dl_layer_mul2d.hpp) for reference.

Recommend Option-1: If you just only have this operation(won't mix with other padding-required operation, such as Conv2D)
`#include <stdint.h>
#include "dl_variable.hpp"
#include "dl_constant.hpp"
#include "dl_nn_mul2d.hpp"

using namespace dl;
using namespace nn;
extern "C" void app_main(void)
{
    int16_t element1[] = {5, 6, 7, 8};
    int16_t element2[] = {1, 2, 3, 4};
    Tensor<int16_t> input1;
    input1.set_element((int16_t *)element1).set_exponent(0).set_shape({2, 2, 1}).set_auto_free(false);
    Tensor<int16_t> input2;
    input2.set_element((int16_t *)element2).set_exponent(0).set_shape({2, 2, 1}).set_auto_free(false);
    Tensor<int16_t> output = dl::nn::mul2d(0, input1, input2, (Activation<int16_t> *)NULL);
    output.print(0, 2, 0, 2, 0, 1, "output", 2);
}`

Recommend Option-2: If this operation is mixed with other padding-required operations, I recommend you dl::layer::Mul2D.

PureHing, 2021-08-11:
Thanks.
BTW, How to achieve the same function as `numpy.dot`

yehangyang, 2021-08-11:
@PureHing 
It's not implemented yet. We planed to support it later.

closed 2021-08-19 by PureHing