espressif/esp-dl #46: Can't use from functions dl_matrixq_from_matrix3d_qmf and dl_matrixq_from_matrix3d from libdl.a

SRavit1, 2021-08-23:
I'm using the esp-dl API to run MNIST on the ESP-EYE board. In my project, I tried using the functions dl_matrixq_from_matrix3d_qmf and dl_matrixq_from_matrix3d.

During linking, I get the following error
~/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: ~/esp-who/components/esp-face/lib/libdl.a(dl_lib_matrix3dq.c.obj): in function `dl_matrixq_from_matrix3d_qmf':
/home/yehangyang/Documents/Gitlab/others/dl_lib/components/dl_lib/dl_lib_matrix3dq.c:61: undefined reference to `rintf'

This error makes me think there's a mistake in the dl_lib_matrix3dq.c file used to make libdl.a. Is this correct? If so, do you have any plans of fixing this in the near future?

Thanks.

yehangyang, 2021-08-24:
Hi @SRavit1,

ESP-DL is facing a thorough refactor. The "dl_matrixq_from_matrix3d_qmf" and "dl_matrixq_from_matrix3d" are in the old version, which will not be supported anymore.

Please, look at the new version in the branch [v2_alpha](https://github.com/espressif/esp-dl/tree/v2_alpha). The branch will be merged into master about this Thursday officially.

Sorry for any inconvenience from this refactor.

Best Regards.

SRavit1, 2021-08-25:
Hi @yehangyang,

Thanks for the reply. I was able to resolve my original issue by manually copying the values from the float matrix to the quantized matrix. 

Thanks for pointing out the new version, I'll have a look.

Regards,
Ravit

closed 2021-08-25 by SRavit1