espressif/esp-dl #47: 调用GlobalAveragePool2D遇到了，如下错误

xiao-mb, 2021-08-24:
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp: In constructor 'dl::layer::GlobalAveragePool2D<feature_t>::GlobalAveragePool2D(int, const char*)':
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp:34:13: error: expected identifier before '{' token
             {
             ^
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp: In member function 'void dl::layer::GlobalAveragePool2D<feature_t>::build(dl::Tensor<feature_t>&)':
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp:54:17: error: 'vector' was not declared in this scope
                 vector<int> output_shape(input.shape.size(), 1);
                 ^~~~~~
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp:54:17: note: suggested alternative:
In file included from c:\users\g\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\8.4.0\vector:64,
                 from ../components/esp-dl/include/tool/dl_tool.hpp:3,
                 from ../main/app_order_recon.cpp:16:
c:\users\g\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\8.4.0\bits\stl_vector.h:339:11: note:   'std::vector'
     class vector : protected _Vector_base<_Tp, _Alloc>
           ^~~~~~
In file included from ../main/includes/order_model.hpp:6,
                 from ../main/app_order_recon.cpp:17:
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp:54:24: error: expected primary-expression before 'int'
                 vector<int> output_shape(input.shape.size(), 1);
                        ^~~
../components/esp-dl/include/layer/dl_layer_global_avg_pool2d.hpp:55:17: error: 'output_shape' was not declared in this scope

yehangyang, 2021-08-24:
Hi @xiao-mb,

Thank you for your feedback. We'll solve it as soon as possible.

yehangyang, 2021-08-24:
Hi @xiao-mb,

Two bugs here.

[/include/layer/dl_layer_global_avg_pool2d.hpp#L33](https://github.com/espressif/esp-dl/blob/cc615c23793efffb5380b32b2c3d3eae54df7e98/include/layer/dl_layer_global_avg_pool2d.hpp#L33):
replace
```
GlobalAveragePool2D(const int output_exponent, const char *name = NULL) : Layer(name),
```
with
```
GlobalAveragePool2D(const int output_exponent, const char *name = NULL) : Layer(name), output_exponent(output_exponent)
```

[include/layer/dl_layer_global_avg_pool2d.hpp#L54](https://github.com/espressif/esp-dl/blob/cc615c23793efffb5380b32b2c3d3eae54df7e98/include/layer/dl_layer_global_avg_pool2d.hpp#L54)：
replace
```
vector<int> output_shape(input.shape.size(), 1);
```
with
```
std::vector<int> output_shape(input.shape.size(), 1);
```

yehangyang, 2021-08-25:
@xiao-mb 如果问题已解决，请关闭 issue。

closed 2021-08-25 by xiao-mb