espressif/esp-dl #48: 关于运行时，conv2d中出现的错误。

xiao-mb, 2021-08-24:
extern "C" uint8_t app_recon(uint8_t *img)
{
    // input
    Tensor<int16_t> input;
    input.set_element((int16_t *)img).set_exponent(0).set_shape({64, 64, 3}).set_auto_free(false);
    ORDER model;
    dl::tool::Latency latency;

    // model forward
    latency.start();
    model.forward(input);
    latency.end();
}

当运行到model.forwad(input);时出现了如下错误：
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was un
handled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x82017c47  A1      : 0x3f
c9ee70
A2      : 0x3dddd010  A3      : 0x3fce1dd0  A4      : 0x00000000  A5      : 0x00
000000
A6      : 0x3fce1d28  A7      : 0x00000000  A8      : 0x82017771  A9      : 0x3f
c9ee50
A10     : 0x3fce1dd0  A11     : 0x3fce25f0  A12     : 0x3fce1d28  A13     : 0x3f
ce25f0
A14     : 0x00000000  A15     : 0x00000042  SAR     : 0x00000004  EXCCAUSE: 0x00
000014
EXCVADDR: 0x00000000  LBEG    : 0x4037d810  LEND    : 0x4037d815  LCOUNT  : 0xff
ffffff
0x4037d810: dl_tie728_bzero at ??:?

0x4037d815: dl_tie728_bzero at ??:?

Backtrace:0xfffffffd:0x3fc9ee700x42017c44:0x3fc9eeb0 0x42016d0c:0x3fc9eed0 0x420
09f69:0x3fc9ef10 0x42016dfd:0x3fc9eff0 0x4200b134:0x3fc9f010 0x4200935a:0x3fc9f5
70 0x4038215d:0x3fc9fc10
0x42017c44: void dl::base::conv2d<short, short, short, short, int>(void*) at /ho
me/yehangyang/Documents/workspace/DL/build/../components/dl/base/dl_base_conv2d.
cpp:895

0x42016d0c: dl::nn::conv2d(dl::Tensor<short>&, dl::Tensor<short>&, std::vector<i
nt, std::allocator<int> >&, dl::Filter<short> const&, int, int, dl::Bias<short>
const*, dl::Activation<short> const*, std::vector<int, std::allocator<int> > con
st&) at /home/yehangyang/Documents/workspace/DL/build/../components/dl/nn/src/dl
_nn_conv2d.cpp:40

0x42009f69: dl::layer::Conv2D<short>::call(dl::Tensor<short>&, bool, std::vector
<int, std::allocator<int> > const&) at C:\Users\gaochao\eclipse-workspace\ESP32S
3_RTOS_SDK_master\build/../components/esp-dl/include/layer/dl_layer_conv2d.hpp:1
15
 (inlined by) ORDER::call(dl::Tensor<short>&) at C:\Users\gaochao\eclipse-worksp
ace\ESP32S3_RTOS_SDK_master\build/../main/includes/order_model.hpp:106

0x42016dfd: dl::layer::Model<short>::forward(dl::Tensor<short>&) at /home/yehang
yang/Documents/workspace/DL/build/../components/dl/layer/src/dl_layer_model.cpp:
30

0x4200b134: app_recon at C:\Users\gaochao\eclipse-workspace\ESP32S3_RTOS_SDK_mas
ter\build/../main/app_order_recon.cpp:31

0x4200935a: task_app_recon_process(void*) at C:\Users\gaochao\eclipse-workspace\
ESP32S3_RTOS_SDK_master\build/../main/app_recon_process.cpp:122

0x4038215d: vPortTaskWrapper at C:/Users/gaochao/Desktop/esp-idf/components/free
rtos/port/xtensa/port.c:168

yehangyang, 2021-08-24:
Hi @xiao-mb,
请展示一下你调用 nn::conv2d(...) 的代码上下文。

xiao-mb, 2021-08-24:
#pragma once
#include "dl_layer_model.hpp"
#include "dl_layer_conv2d.hpp"
#include "dl_layer_depthwise_conv2d.hpp"
#include "dl_layer_concat2d.hpp"
#include "order_coefficient.hpp"
#include <stdint.h>

using namespace dl;
using namespace layer;
using namespace order_coefficient;

class ORDER : public Model<int16_t> // Derive the Model class in "dl_layer_model.hpp"
{
private:
    // Declare layers as member variables
    Conv2D<int16_t> l1_conv;                  // a layer named l1
    DepthwiseConv2D<int16_t> l2_dp;   // a layer named l2_depth
    Conv2D<int16_t> l2_exp;         // a layer named l2_compress
    DepthwiseConv2D<int16_t> l3_dp; // a layer named l3_a_depth
    Conv2D<int16_t> l3_exp;       // a layer named l3_a_compress
    DepthwiseConv2D<int16_t> l4_dp; // a layer named l3_b_depth
    Conv2D<int16_t> l4_exp;       // a layer named l3_b_compress
    DepthwiseConv2D<int16_t> l5_dp; // a layer named l3_c_depth
    Conv2D<int16_t> l5_exp;       // a layer named l3_c_compress
    DepthwiseConv2D<int16_t> l6_dp; // a layer named l3_d_depth
    Conv2D<int16_t> l6_exp;       // a layer named l3_d_compress
    Conv2D<int16_t> l7_conv;       // a layer named l3_e_compress

public:
    Conv2D<int16_t> l8_cp; // a layer named l5_compress

    /**
     * @brief Initialize layers in constructor function
     *
     */
    ORDER() : l1_conv(Conv2D<int16_t>(-16, get_l1_conv_filter(), NULL, get_l1_conv_activation(), PADDING_VALID, 2, 2, "l1")),
              l2_dp(DepthwiseConv2D<int16_t>(-15, get_l2_dp_filter(), NULL, get_l2_dp_activation(), PADDING_SAME, 1, 1, "l2_depth")),
              l2_exp(Conv2D<int16_t>(-3, get_l2_dp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l2_compress")),
              l3_dp(DepthwiseConv2D<int16_t>(-16, get_l3_dp_filter(), NULL, get_l3_dp_activation(), PADDING_SAME, 1, 1, "l3depth")),
              l3_exp(Conv2D<int16_t>(-13, get_l3_exp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l3_a_compress")),
              l4_dp(DepthwiseConv2D<int16_t>(-15, get_l3_dp_filter(), NULL, get_l3_dp_activation(), PADDING_SAME, 2, 2, "l4_depth")),
              l4_exp(Conv2D<int16_t>(-14, get_l4_exp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l3_b_compress")),
              l5_dp(DepthwiseConv2D<int16_t>(-15, get_l4_dp_filter(), NULL, get_l5_dp_activation(), PADDING_SAME, 1, 1, "l5_depth")),
              l5_exp(Conv2D<int16_t>(-13, get_l5_exp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l5_compress")),
              l6_dp(DepthwiseConv2D<int16_t>(-15, get_l6_dp_filter(), NULL, get_l6_dp_activation(), PADDING_SAME, 2, 2, "l6_depth")),
              l6_exp(Conv2D<int16_t>(-15, get_l6_exp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l6_compress")),
              l7_conv(Conv2D<int16_t>(-14, get_l7_conv_filter(), NULL, get_l7_conv_activation(), PADDING_VALID, 1, 1, "l7_compress")),
              l8_cp(Conv2D<int16_t>(-14, get_l8_full_c_filter(), get_l8_full_c_bias(), NULL, PADDING_VALID, 1, 1, "l8_compress")){}

    /**
     * @brief call each layers' build(...) function in sequence
     *
     * @param input
     */
    void build(Tensor<int16_t> &input)
    {
        this->l1_conv.build(input);
        this->l2_dp.build(this->l1_conv.output);
        this->l2_exp.build(this->l2_dp.output);
        this->l3_dp.build(this->l2_exp.output);
        this->l3_exp.build(this->l3_dp.output);
        this->l4_dp.build(this->l3_exp.output);
        this->l4_exp.build(this->l4_dp.output);
        this->l5_dp.build(this->l4_exp.output);
        this->l5_exp.build(this->l5_dp.output);
        this->l6_dp.build(this->l5_exp.output);
        this->l6_exp.build(this->l6_dp.output);
        this->l7_conv.build(this->l6_exp.output);
        this->l8_cp.build(this->l7_conv.output);
    }

    /**
     * @brief call each layers' call(...) function in sequence
     *
     * @param input
     */
    void call(Tensor<int16_t> &input)
    {
        this->l1_conv.call(input);
        input.free_element();

        this->l2_dp.call(this->l1_conv.output);
        this->l1_conv.output.free_element();

        this->l2_exp.call(this->l2_dp.output);
        this->l2_dp.output.free_element();

        this->l3_dp.call(this->l2_exp.output);
        this->l2_exp.output.free_element();

        this->l3_exp.call(this->l3_dp.output);
        this->l3_dp.output.free_element();

        this->l4_dp.call(this->l3_exp.output);
        this->l3_exp.output.free_element();

        this->l4_exp.call(this->l4_dp.output);
        this->l4_dp.output.free_element();

        this->l5_dp.call(this->l4_exp.output);
        this->l4_exp.output.free_element();

        this->l5_exp.call(this->l5_dp.output);
        this->l5_dp.output.free_element();

        this->l6_dp.call(this->l5_exp.output);
        this->l5_exp.output.free_element();

        this->l6_exp.call(this->l6_dp.output);
        this->l6_dp.output.free_element();

        this->l7_conv.call(this->l6_exp.output);
        this->l6_exp.output.free_element();
        this->l8_cp.call(this->l7_conv.output);
        this->l7_conv.output.free_element();
    }
};

yehangyang, 2021-08-25:
函数 uint8_t app_recon(uint8_t *img) 的 img 是如何得到的？

请务必用 [`inline void *malloc_aligned(int number, int size, int align = 0)`](https://github.com/espressif/esp-dl/blob/cc615c23793efffb5380b32b2c3d3eae54df7e98/include/tool/dl_tool.hpp#L77) 申请内存。并且设定 `align = 16`。

xiao-mb, 2021-08-25:
之前是用这个函数申请的内存：uint8_t *img6464 = (uint8_t *)heap_caps_calloc(1, 64 * 64 * 3 * sizeof(uint8_t), MALLOC_CAP_INTERNAL);
换成：uint8_t *img6464 = (uint8_t *)dl::tool::malloc_aligned(1, 64 * 64 * 3 * sizeof(uint8_t), 16);后运行时还是有错误，如下：
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was un
handled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x82017cab  A1      : 0x3f
c9ee70
A2      : 0x3dddd010  A3      : 0x3fce4e0c  A4      : 0x00000000  A5      : 0x00
000000
A6      : 0x3fce4d64  A7      : 0x00000000  A8      : 0x820177d5  A9      : 0x3f
c9ee50
A10     : 0x3fce4e0c  A11     : 0x3fce5630  A12     : 0x3fce4d64  A13     : 0x3f
ce5630
A14     : 0x00000000  A15     : 0x00000042  SAR     : 0x00000004  EXCCAUSE: 0x00
000014
EXCVADDR: 0x00000000  LBEG    : 0x4037d810  LEND    : 0x4037d815  LCOUNT  : 0xff
ffffff
0x4037d810: dl_tie728_bzero at ??:?

0x4037d815: dl_tie728_bzero at ??:?

Backtrace:0xfffffffd:0x3fc9ee700x42017ca8:0x3fc9eeb0 0x42016d70:0x3fc9eed0 0x420
09fc1:0x3fc9ef10 0x42016e61:0x3fc9eff0 0x4200b198:0x3fc9f010 0x420093b2:0x3fc9f5
70 0x4038215d:0x3fc9fc10
0x42017ca8: void dl::base::conv2d<short, short, short, short, int>(void*) at /ho
me/yehangyang/Documents/workspace/DL/build/../components/dl/base/dl_base_conv2d.
cpp:895

0x42016d70: dl::nn::conv2d(dl::Tensor<short>&, dl::Tensor<short>&, std::vector<i
nt, std::allocator<int> >&, dl::Filter<short> const&, int, int, dl::Bias<short>
const*, dl::Activation<short> const*, std::vector<int, std::allocator<int> > con
st&) at /home/yehangyang/Documents/workspace/DL/build/../components/dl/nn/src/dl
_nn_conv2d.cpp:40

0x42009fc1: dl::layer::Conv2D<short>::call(dl::Tensor<short>&, bool, std::vector
<int, std::allocator<int> > const&) at C:\Users\gaochao\eclipse-workspace\ESP32S
3_RTOS_SDK_master\build/../components/esp-dl/include/layer/dl_layer_conv2d.hpp:1
15
 (inlined by) ORDER::call(dl::Tensor<short>&) at C:\Users\gaochao\eclipse-worksp
ace\ESP32S3_RTOS_SDK_master\build/../main/includes/order_model.hpp:107

0x42016e61: dl::layer::Model<short>::forward(dl::Tensor<short>&) at /home/yehang
yang/Documents/workspace/DL/build/../components/dl/layer/src/dl_layer_model.cpp:
30

0x4200b198: app_recon at C:\Users\gaochao\eclipse-workspace\ESP32S3_RTOS_SDK_mas
ter\build/../main/app_order_recon.cpp:31

0x420093b2: task_app_recon_process(void*) at C:\Users\gaochao\eclipse-workspace\
ESP32S3_RTOS_SDK_master\build/../main/app_recon_process.cpp:123

0x4038215d: vPortTaskWrapper at C:/Users/gaochao/Desktop/esp-idf/components/free
rtos/port/xtensa/port.c:168

yehangyang, 2021-08-25:
uint8_t *img6464 = (uint8_t *)dl::tool::malloc_aligned(64 * 64 * 3, sizeof(uint8_t), 16);

xiao-mb, 2021-08-25:
修改之后，还是同样的错误

yehangyang, 2021-08-25:
```
(inlined by) ORDER::call(dl::Tensor&) at C:\Users\gaochao\eclipse-worksp
ace\ESP32S3_RTOS_SDK_master\build/../main/includes/order_model.hpp:107
```
这一行指向的哪一行代码？

xiao-mb, 2021-08-25:
this->l3_exp.call(this->l3_dp.output);

yehangyang, 2021-08-25:
typo here:
```
          l3_dp(DepthwiseConv2D<int16_t>(-16, get_l3_dp_filter(), NULL, get_l3_dp_activation(), PADDING_SAME, 1, 1, "l3depth")),
          l3_exp(Conv2D<int16_t>(-13, get_l3_exp_filter(), NULL, NULL, PADDING_SAME, 1, 1, "l3_a_compress")),
          l4_dp(DepthwiseConv2D<int16_t>(-15, **get_l3_dp_filter()**, NULL, **get_l3_dp_activation()**, PADDING_SAME, 2, 2, "l4_depth")),
```

yehangyang, 2021-08-25:
@xiao-mb 如果问题已解决，请关闭 issue。

closed 2021-08-25 by xiao-mb