espressif/esp-dl #49: 关于conv2d类型的错误

xiao-mb, 2021-08-24:
l7_gpool("l7_global_pool"),
l8_cp(Conv2D<int16_t>(-14, get_l8_full_c_filter(), get_l8_full_c_bias(), NULL, PADDING_VALID, 1, 1, "l8_compress")){}

编译上面的代码时，显示类型的错误如下：
In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp: In constructor 'ORDER::ORDER()':
../main/includes/order_model.hpp:53:129: error: invalid conversion from 'const char*' to 'int' [-fpermissive]
               l8_cp(Conv2D<int16_t>(-14, get_l8_full_c_filter(), get_l8_full_c_bias(), NULL, PADDING_VALID, 1, 1, "l8_compress")){}

但是没找到哪里类型不对

yehangyang, 2021-08-25:
Hi @xiao-mb,

l7_gpool(**[缺少输出 tensor 的 exponent 设定]**,"l7_global_pool"),

xiao-mb, 2021-08-25:
[缺少输出 tensor 的 exponent 设定]，这一层没有参数，这个exponent要怎么确定？

yehangyang, 2021-08-25:
与参数无关，这是输出 tensor 的 exponent。

xiao-mb, 2021-08-25:
这个输出tensor，应该是上一次的输出经过池化后得到的，与输入有关系，不同的输入图片，max(abs(value_float))是否也会不同，如何确定max(abs(value_float))？

yehangyang, 2021-08-25:
Hi @xiao-mb,

所有操作（除了池化，还有卷积等）的输出 tensor 的 exponent 都需要根据实际输出（浮点型）+ 公式`exponent = log2(max(abs(value_float)) / 2^(element_width - 1))`去确定。不同输入图片确实会带来不同的输出 tensor 的 exponent。所以在确定输出 tensor 的 exponent 时，可以用 batch 稍大的图片去做为输入，得到统计意义的 max(abs(value_float))。

closed 2021-08-25 by xiao-mb