espressif/esp-dl #52: 运行esp-dl/tools/quantization_tool/examples/example.py时出现的错误

xiao-mb, 2021-08-31:
当运行esp-dl/tools/quantization_tool/examples/example.py时，出现如下错误，我的平台是windows
 File "D：/esp-dl/tools/quantization_tool/examples/example.py", line 16, in <module>
    from calibrator import *
ModuleNotFoundError: No module named 'calibrator'

Auroragan, 2021-08-31:
目前暂时还不支持 windows 平台, 只支持 linux

xiao-mb, 2021-08-31:
windows 平台计划什么时候支持？

Auroragan, 2021-08-31:
预计这周会推上去

Auroragan, 2021-09-03:
请查看最新的 master 分支

xiao-mb, 2021-09-06:
拉去最新master分支，还是同样的错误

Auroragan, 2021-09-06:
请问你的windows 是哪个版本?

xiao-mb, 2021-09-06:
win10 专业版  版本号：21H1

Auroragan, 2021-09-06:
```
path = f'../{system_type.lower()}'
if system_type == 'Windows':
    path = path.replace('/', '\\')
sys.path.append(path)
```

这里的 path 你那边打印出来的是什么?

xiao-mb, 2021-09-06:
..\windows

xiao-mb, 2021-09-07:
装了Linux虚拟机运行后，发生了如下错误：
gc@gc-VirtualBox:/mnt/esp-dl/tools/quantization_tool/examples$ python3 example.py
<frozen importlib._bootstrap>:219: RuntimeWarning: compiletime version 3.7 of module 'calibrator' does not match runtime version 3.8
<frozen importlib._bootstrap>:219: RuntimeWarning: compiletime version 3.7 of module 'evaluator' does not match runtime version 3.8
Generating the quantization table:
Exporting finish, the output files are: ./test_mnist.cpp, ./test_mnist.hpp

Quantized model info:
model input name: input, exponent: -15
Reshape layer name: sequential/flatten/Reshape, output_exponent: -15
Gemm layer name: fused_gemm_0, output_exponent: -11
Gemm layer name: fused_gemm_1, output_exponent: -11
Gemm layer name: fused_gemm_2, output_exponent: -9

**Evaluating the performance on esp32s3:
Traceback (most recent call last):
  File "example.py", line 44, in <module>
    eva.generate_quantized_model(model_proto, pickle_file_path)
  File "evaluator.py", line 526, in evaluator.Evaluator.generate_quantized_model
  File "evaluator.py", line 791, in evaluator.Evaluator.generate_quantized_model_s16
AttributeError: 'google.protobuf.pyext._message.RepeatedCompositeCo' object has no attribute 'append'**

python版本改为3.7后又发出现了如下错误：
gc@gc-VirtualBox:/mnt/esp-dl/tools/quantization_tool/examples$ python example.py
Traceback (most recent call last):
  File "/home/gc/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/gc/.local/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/gc/.local/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ModuleNotFoundError: No module named 'numpy.core._multiarray_umath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "example.py", line 1, in <module>
    import numpy as np
  File "/home/gc/.local/lib/python3.7/site-packages/numpy/__init__.py", line 150, in <module>
    from . import core
  File "/home/gc/.local/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/usr/bin/python"
  * The NumPy version is: "1.21.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'

如何才能搭建一个可以运行的环境？能够提供下python numpy 版本？

Auroragan, 2021-09-07:
win10 的环境我们正在调试中

你用的是 conda 环境吗? 是用 pip install -r requirement.txt 安装的依赖包吗? 
我这边测试了一下 python3.7, numpy 1.21.2 是可以跑的

xiao-mb, 2021-09-07:
不是用的conda环境，单独安装的依赖包。我再试下conda环境

xiao-mb, 2021-09-07:
重装虚拟机，用conda环境试下来是可以的。

closed 2021-09-07 by xiao-mb