espressif/esp-dl #57: final link failed: bad value

HeloWong, 2021-09-07:
when I  use function **math/matrix::get_perspective_transform(src, dst)** on **windows10  ESP-IDF v4.1.2**, I got a compile error:
**final link failed: bad value**

```
cmd.exe /C "cd . && C:\Users\Wong\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address -g   @CMakeFiles\hello_dl.elf.rsp  -o hello_dl.elf  && cd ."
c:/users/wong/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
and my part code : 

```C
#include <stdio.h>
#include <stdlib.h>
#include <vector>

#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "dl_math_matrix.hpp"
#include "dl_tool.hpp"

extern "C" void app_main(void)
{
    dl::math::Matrix<float> src(4, 2);
    dl::math::Matrix<float> dst(4, 2);
    src.set_random(255);
    dst.set_random(255);
  
    src.print_value();
    dst.print_value();
    dl::math::Matrix<float> t = dl::math::get_perspective_transform(src, dst);
    t.print_value();
}
```

How should I do it with dl?

igrr, 2021-09-07:
@yehangyang FYI, this issue happens because object files produced by binutils 2.35.1 (used in toolchain esp-2021r1) are incompatible with the linker in binutils 2.31.1 (used in earlier versions of the toolchain) because of new relocation types introduced in 2.35. More details and possible solutions are available in internal Jira: GCC-150.

yehangyang, 2021-09-08:
Hi @HeloWong I'll check it as soon as possible.

Hi @igrr , thanks, I'll check the GCC-150.

yehangyang, 2021-09-08:
Hi @HeloWong,
Would you mind changing to the newest ESP-IDF/master temporarily? Please let me know if any other problem happens.

HeloWong, 2021-09-08:
Thanks @igrr @yehangyang . It works, when I change to toolchain (esp-2021r1) from (esp-2020r3). But I cannot do it on development  machine provisionally. by the way, is it possbile to have plan to release source code?

yehangyang, 2021-09-08:
Sorry @HeloWong , we have no plan to release the source code of ESP-DL.

closed 2021-09-08 by HeloWong