espressif/esp-dl #58: 用量化工具时遇到了current model is not supported by esp-dl的错误

xiao-mb, 2021-09-08:
Generating the quantization table:
MatMul is not supported on esp-dl yet
Softmax is not supported on esp-dl yet
Traceback (most recent call last):
  File "example.py", line 40, in <module>
    calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)
  File "calibrator.py", line 320, in calibrator.Calibrator.generate_quantization_table
  File "calibrator.py", line 209, in calibrator.Calibrator.generate_output_model
  File "calibrator.py", line 204, in calibrator.Calibrator.check_model
ValueError: current model is not supported by esp-dl
用量化工具时遇到了以上错误，要如何修改？

Auroragan, 2021-09-08:
因为 esp-dl 的 API 暂不支持 matmul 和 softmax, 所以 calibrator 暂不能接受这个模型

xiao-mb, 2021-09-08:
看例子里的Dense层被优化成Gemm，我模型里的Dense还是matmul，没有被优化为Gemm，这是为什么？

Auroragan, 2021-09-08:
你这一层没有 bias 吗?

xiao-mb, 2021-09-08:
有的

Auroragan, 2021-09-08:
能把模型图发出来看一下吗?

xiao-mb, 2021-09-08:
[image]
上图是优化前的：
下图是优化后的：
[image]

Auroragan, 2021-09-08:
试试把 optimizer.py 
https://github.com/espressif/esp-dl/blob/8a5be0973686e8ae16e19a345025a954b5cdc31e/tools/quantization_tool/optimizer.py#L137
```
if not dim[0].dim_param:
    dim[0].dim_param = 'N'
```

改成 固定的batch size:
```
dim[0].dim_param = '20'
```

看一下能不能 fuse 成功

xiao-mb, 2021-09-08:
[image]
还是不行

Auroragan, 2021-09-09:
dim[0].dim_param = ''
dim[0].dim_value = 20

改成这样呢?

xiao-mb, 2021-09-09:
可以了。能生成模型文件的版本麻烦尽快更新下，谢谢！

Auroragan, 2021-09-10:
我们 API 将会有所改动, 能生成模型文件的版本将在之后与 API 一起更新

xiao-mb, 2021-09-12:
> dim[0].dim_param = ''
> dim[0].dim_value = 20
> 
> 改成这样呢?

batch size 的大小有什么讲究吗，把网络移植到s3上后，识别不正确。评估的准确率如下：

> Evaluating the performance on esp32s3:
>accuracy of int16 model is: 0.996094
>accuracy of fp32 model is: 0.996094
把同张图片导出，在PC上测试，识别正确。在S3上，不同的图片大多数会预测出同一个值。这个问题应该从哪些方面来定位原因？