espressif/esp-dl #59: IndexError: list index (1) out of range--------calibrator.Calibrator.export_coefficient_to_cpp

PureHing, 2021-09-08:
my onnx model input size is bx240x320x3，calib_dataset shape is (36,240,320,3)
<details>
  <summary markdown="span">log</summary>

```
tools/quantization_tool/examples$ python example.py 
calib_dataset shape: (36, 240, 320, 3)
Generating the quantization table:
Traceback (most recent call last):
  File "example.py", line 48, in <module>
    calib.export_coefficient_to_cpp(model_proto, pickle_file_path, 'esp32s3', '.', 'test_mnist', True)
  File "calibrator.py", line 589, in calibrator.Calibrator.export_coefficient_to_cpp
IndexError: list index (1) out of range
```

</details>

What's the problem？

PureHing, 2021-09-09:
example.py code:
```

if __name__ == "__main__":
    # Optimize the onnx model
    model_path = 'best_tflite.onnx'#"mnist_model_example.onnx"#
    optimized_model_path = optimize_fp_model(model_path)

    # Calibration
    with open('test.pickle', 'rb') as f:#mnist_test_data.pickle
        test_images = pickle5.load(f)
        # test_images = pickle.load(f)
    # test_images = test_images / 255.0

    # Prepare the calibration dataset
    calib_dataset = test_images#[0:5000:50]
    print("calib_dataset shape:",calib_dataset.shape)
    pickle_file_path = 'alib.pickle'
    model_proto = onnx.load(optimized_model_path)

    print('Generating the quantization table:')
    calib = Calibrator('int16', 'per-tensor', 'minmax')
    calib.set_providers(['CPUExecutionProvider'])
    calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)
    calib.export_coefficient_to_cpp(model_proto, pickle_file_path, 'esp32s3', '.', 'test_mnist', True)
```

Auroragan, 2021-09-09:
Hi @PureHing 
what is the activation function for this layer?

not sure what you think as more friendly, both depthwise_conv2d and conv2d are optimized on S3. you can choose based on your application.

PureHing, 2021-09-09:
@Auroragan According to the error prompt:
```
tools/quantization_tool/examples$ python example.py 
calib_dataset shape: (36, 240, 320, 3)
Generating the quantization table:
Traceback (most recent call last):
  File "example.py", line 48, in <module>
    calib.export_coefficient_to_cpp(model_proto, pickle_file_path, 'esp32s3', '.', 'test_best', True)
  File "calibrator.py", line 589, in calibrator.Calibrator.export_coefficient_to_cpp
IndexError: list index (1) out of range

```
I don't know which list is out of range.
All activation is leakyrelu.

Auroragan, 2021-09-09:
yeah ok, it's a bug for exporting leakyrelu activation, will fix it soon

PureHing, 2021-09-09:
@Auroragan ok.
I changed the activation to relu ,and excuted `python example.py`,get the follow results:

```
Quantized model info:
model input name: input_1:0, exponent: -15
Reshape layer name: model_conv2d_Conv2D__10, output_exponent: -15
Conv layer name: model_conv2d_Conv2D, output_exponent: -11
MaxPool layer name: model_max_pooling2d_MaxPool, output_exponent: -11
Conv layer name: model_conv2d_1_Conv2D, output_exponent: -11
Conv layer name: model_conv2d_2_Conv2D, output_exponent: -11
Concat layer name: model_concatenate_concat, output_exponent: -11
Conv layer name: model_conv2d_3_Conv2D, output_exponent: -11
MaxPool layer name: model_max_pooling2d_1_MaxPool, output_exponent: -11
Conv layer name: model_conv2d_4_Conv2D, output_exponent: -12
MaxPool layer name: model_max_pooling2d_2_MaxPool, output_exponent: -12
Conv layer name: model_conv2d_5_Conv2D, output_exponent: -12
Conv layer name: model_conv2d_6_Conv2D, output_exponent: -11
Conv layer name: model_conv2d_7_Conv2D, output_exponent: -11
Conv layer name: model_conv2d_8_Conv2D, output_exponent: -11
MaxPool layer name: model_max_pooling2d_3_MaxPool, output_exponent: -11
Conv layer name: model_conv2d_9_Conv2D, output_exponent: -11
Conv layer name: model_conv2d_10_Conv2D, output_exponent: -11
Conv layer name: model_conv2d_11_Conv2D, output_exponent: -10
Conv layer name: model_conv2d_12_BiasAdd, output_exponent: -10
Transpose layer name: model_conv2d_12_BiasAdd__120, output_exponent: -10
```

and generated the hpp file by  `export_coefficient_to_cpp` function :

<details>
  <summary markdown="span">hpp file</summary>

```
#pragma once

#include <stdint.h>
#include "dl_constant.hpp"

namespace test_dm
{
    const dl::Filter<int16_t> *get_model_conv2d_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_1_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_1_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_1_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_2_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_2_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_2_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_3_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_3_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_3_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_4_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_4_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_4_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_5_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_5_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_5_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_6_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_6_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_6_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_7_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_7_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_7_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_8_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_8_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_8_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_9_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_9_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_9_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_10_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_10_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_10_Conv2D_activation();

    const dl::Filter<int16_t> *get_model_conv2d_11_Conv2D_filter();
    const dl::Bias<int16_t> *get_model_conv2d_11_Conv2D_bias();
    const dl::Activation<int16_t> *get_model_conv2d_11_Conv2D_activation();
    
    const dl::Filter<int16_t> *get_model_conv2d_12_BiasAdd_filter();
    const dl::Bias<int16_t> *get_model_conv2d_12_BiasAdd_bias();
}

```

</details>

**question:**
After I got the quantization parameter hpp and cpp file, I refer to [mnist_model.hpp] (https://github.com/espressif/esp-dl/blob/master/tutorial/model/mnist_model.hpp) to generate my own model.hpp, how should I declare the Reshape layer and Transpose layer

TiramisuJ, 2021-09-10:
@PureHing Yeah，It‘s right. 
Sorry that the current version does not support Reshape layer and Transpose layer. and we will update them soon in the next version.

PureHing, 2021-09-10:
Maybe I can cut off the input and output of reshape and transpose layer, because only dimensionality conversion is done here.

TiramisuJ, 2021-09-10:
Sure，the image channel conversion from 320x320x1 to 1x320x320 doesn’t change the data arangement,  you can just change the shape of the Tensor from  [320, 320, 1] to [1, 320, 320].  
Please wait a minute, we will fix some bugs and push a new version tonight. please use the new version to develop your project.

TiramisuJ, 2021-09-10:
@PureHing  The new version has been pushed. you can try it

PureHing, 2021-09-10:
@TiramisuJ @Auroragan the newest version didn't have anychanges for my model.
the all activation of my model is relu now, so if I want to evaluate this model on esp32s3 now, the reshape layer and transpose layer just can deal with what I have said in [here.](https://github.com/espressif/esp-dl/issues/59#issuecomment-916652396)

Auroragan, 2021-09-10:
There should be some differences about the value in the exported cpp? 

Yes you can cut off the reshape and transpose layer. For the input, you don’t need to do any change, because for our library the input dimensions for conv should be [height, width, channel], so if you have the array of image data for example:
__attribute__((aligned(16))) int16_t test_image[320*320]={...}

Tensor<int16_t> input;
input.set_element((int16_t *)test_image).set_exponent(*).set_shape({320,320,1}).set_auto_free(false);

For the output, the dimensions are also in the same sequence as above

PureHing, 2021-09-10:
@Auroragan  Thanks much. Got it!

PureHing, 2021-09-10:
@Auroragan BTW, How is the `dst_h` parameter given [here](https://github.com/espressif/esp-dl/blob/dcaaa939edeef575a7340ff28dcfe8ff1411b988/include/image/dl_image.hpp#L156)

Auroragan, 2021-09-11:
You don’t need to specify dst_height because it’s scaling the width and height with an equal ratio

PureHing, 2021-09-11:
@Auroragan 
ok.
From this [demo](https://github.com/espressif/esp-dl/blob/dcaaa939edeef575a7340ff28dcfe8ff1411b988/examples/human_face_detect/main/app_main.cpp#L25), Is the quantization method of the face detect model obtained through int8 quantization,and  the input element is uint8.
Here is how to do the conversion？my model is calibrated by using `Calibrator('int8', 'per-tensor', 'entropy')`

Auroragan, 2021-09-11:
Our face detect model is using 16-bit quantization, don’t need to convert the input type

if you want to use 8-bit quantization, you can normalized your input image first as what you did for your original model and then times 2^-input_exponent to be the int8 input tensor, or you can also simplify this calculation based on the normalization method you used

PureHing, 2021-09-11:
ok.
Can you evaluate the time cost on esp32s3 in totally with the model which input is 320x320x1 and 5 times downsample and all operator is convolutional?

Auroragan, 2021-09-11:
sorry cannot evaluate the exact time.

> forward: 11205758 us
is this the inference time of running your model on S3?

PureHing, 2021-09-11:
yes. on s3

Auroragan, 2021-09-11:
doesn't look correct, your model is not heavy. channel = 256 is not a problem. For our 8-bit face recognition model, the inference time is only 280ms.

What is your configuration for cache size, cache line size?

PureHing, 2021-09-11:
```
#
# Cache config
#
# CONFIG_ESP32S3_INSTRUCTION_CACHE_16KB is not set
CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB=y
CONFIG_ESP32S3_INSTRUCTION_CACHE_SIZE=0x8000
# CONFIG_ESP32S3_INSTRUCTION_CACHE_4WAYS is not set
CONFIG_ESP32S3_INSTRUCTION_CACHE_8WAYS=y
CONFIG_ESP32S3_ICACHE_ASSOCIATED_WAYS=8
CONFIG_ESP32S3_INSTRUCTION_CACHE_LINE_32B=y
CONFIG_ESP32S3_INSTRUCTION_CACHE_LINE_SIZE=32
CONFIG_ESP32S3_INSTRUCTION_CACHE_WRAP=y
# CONFIG_ESP32S3_DATA_CACHE_16KB is not set
# CONFIG_ESP32S3_DATA_CACHE_32KB is not set
CONFIG_ESP32S3_DATA_CACHE_64KB=y
CONFIG_ESP32S3_DATA_CACHE_SIZE=0x10000
# CONFIG_ESP32S3_DATA_CACHE_4WAYS is not set
CONFIG_ESP32S3_DATA_CACHE_8WAYS=y
CONFIG_ESP32S3_DCACHE_ASSOCIATED_WAYS=8
CONFIG_ESP32S3_DATA_CACHE_LINE_32B=y
CONFIG_ESP32S3_DATA_CACHE_LINE_SIZE=32
CONFIG_ESP32S3_DATA_CACHE_WRAP=y
# end of Cache config
```

PureHing, 2021-09-13:
> > In this model:Layer 7---model/conv2d_7/Conv2D: bx64x20x20->bx128x20x20,it cost about 1150ms.
> 
> is it 3*3 convolution? Let me test it on the work day.

@yehangyang Hi,I'm sorry to bother you,Have you tested it?

yehangyang, 2021-09-13:
@PureHing Sorry, I'm on leave today. I'll test it tomorrow.

Auroragan, 2021-09-13:
@PureHing  I've tested the model you provided before, the inference time is similar to your result, I guess you need to redesign the model structure if you want a faster model.

PureHing, 2021-09-13:
@Auroragan Thanks
That is to use `dw conv2d` as much as possible, is the input( 320x320x1 )  so large?

Auroragan, 2021-09-13:
I guess, the size of tensors from the first few layers are like 400-800 kB, it's pretty large

PureHing, 2021-09-13:
@Auroragan 
HI

I calculated that the first few layers of the model are only about 300-400ms.

It consumes more than 2,000 ms since the conv2d_7 layer in the model.

Auroragan, 2021-09-13:
they are not conflict. ok it's like this:
first, the size of first few layers are relatively large, I don't know if it meets your requirements, but i think 300-400ms is still not fast for running a few layer
secondly, it takes long time for that layer because the MAC times is large, so maybe you can use 1*1 conv instead

PureHing, 2021-09-14:
ok, thanks much

closed 2021-09-14 by PureHing