espressif/esp-dl #60: 移植过程中遇到了编译错误dl::Tensor<short int>* dl::layer::Conv2D<short int>::output' is private within this context

xiao-mb, 2021-09-09:
log如下：

 In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp: In member function 'virtual void ORDER::build(dl::Tensor<short int>&)':
../main/includes/order_model.hpp:69:41: error: 'dl::Tensor<short int>* dl::layer::Conv2D<short int>::output' is private within this context
         this->l2_dp.build(this->l1_conv.output);
                                         ^~~~~~
In file included from ../main/includes/order_model.hpp:3,
                 from ../main/app_order_recon.cpp:17:
../components/esp-dl/include/layer/dl_layer_conv2d.hpp:29:32: note: declared private here
             Tensor<feature_t> *output;              /*<! output ptr of Conv2D >*/
                                ^~~~~~
In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp:69:47: error: no matching function for call to 'dl::layer::DepthwiseConv2D<short int>::build(dl::Tensor<short int>*&)'
         this->l2_dp.build(this->l1_conv.output);
                                               ^
In file included from ../main/includes/order_model.hpp:4,
                 from ../main/app_order_recon.cpp:17:
../components/esp-dl/include/layer/dl_layer_depthwise_conv2d.hpp:86:18: note: candidate: 'void dl::layer::DepthwiseConv2D<feature_t>::build(dl::Tensor<feature_t>&) [with feature_t = short int]'
             void build(Tensor<feature_t> &input)
                  ^~~~~
../components/esp-dl/include/layer/dl_layer_depthwise_conv2d.hpp:86:18: note:   no known conversion for argument 1 from 'dl::Tensor<short int>*' to 'dl::Tensor<short int>&'
In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp:70:40: error: 'dl::Tensor<short int>* dl::layer::DepthwiseConv2D<short int>::output' is private within this context
         this->l2_exp.build(this->l2_dp.output);
                                        ^~~~~~
In file included from ../main/includes/order_model.hpp:4,
                 from ../main/app_order_recon.cpp:17:
../components/esp-dl/include/layer/dl_layer_depthwise_conv2d.hpp:29:32: note: declared private here
             Tensor<feature_t> *output;               /*<! output ptr of DepthwiseConv2D >*/
                                ^~~~~~
In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp:70:46: error: no matching function for call to 'dl::layer::Conv2D<short int>::build(dl::Tensor<short int>*&)'
         this->l2_exp.build(this->l2_dp.output);
                                              ^
In file included from ../main/includes/order_model.hpp:3,
                 from ../main/app_order_recon.cpp:17:
../components/esp-dl/include/layer/dl_layer_conv2d.hpp:86:18: note: candidate: 'void dl::layer::Conv2D<feature_t>::build(dl::Tensor<feature_t>&) [with feature_t = short int]'
             void build(Tensor<feature_t> &input)
                  ^~~~~
../components/esp-dl/include/layer/dl_layer_conv2d.hpp:86:18: note:   no known conversion for argument 1 from 'dl::Tensor<short int>*' to 'dl::Tensor<short int>&'
In file included from ../main/app_order_recon.cpp:17:
../main/includes/order_model.hpp:71:40: error: 'dl::Tensor<short int>* dl::layer::Conv2D<short int>::output' is private within this context
         this->l3_dp.build(this->l2_exp.output);
                                        ^~~~~~

xiao-mb, 2021-09-09:
同样的代码，在v2_alpha版本上可以正常编译

TiramisuJ, 2021-09-10:
Hi， 你好， 我们在1.0版本调整了接口， 每个layer的output请使用他们的成员函数get_output()来获得。

closed 2021-09-10 by xiao-mb