espressif/esp-dl #61: Squeeze层在s3上如何实现？

xiao-mb, 2021-09-10:
[image]
如上图，squeeze层在s3上如何实现？

xiao-mb, 2021-09-10:
因为Conv层的输出是60x1x1，所以目前在s3上跳过了GlobalAveragePool与Squeeze这2个层的实现，直接接的Gemm层，Gemm是用的1x1的卷积，但在s3上测试效果与pc上差了很多。不知道是不是这2个层移植的问题？

TiramisuJ, 2021-09-10:
@xiao-mb GlobalAveragePool在s3上可以实现，Squeeze 和 Gemm层可以用一个1x1卷积代替， 如果你的实现是正确的， 16bit模型的效果应该和PC端相似，8bit模型可能会略有下降。

xiao-mb, 2021-09-10:
在s3上增加了GlobalAveragePool层，代码如下：

      Conv2D<int16_t> l9_exp;   				// a layer named l9_exp
      GlobalAveragePool2D<int16_t> l9_gap; 
      

        l9_exp(Conv2D<int16_t>(-12, get_test_model_l9_exp_Conv2D_filter(), get_test_model_l9_exp_Conv2D_bias(), NULL, PADDING_VALID, 1, 1, "l9_compress")),
        l9_gap(GlobalAveragePool2D<int16_t>(-12, "l9_GAP")),
        l10_cp(Conv2D<int16_t>(-9, get_fused_gemm_0_filter(), get_fused_gemm_0_bias(), NULL, PADDING_VALID, 1, 1, "l10_compress")){}

        this->l9_exp.build(this->l9_dp.get_output());
        this->l9_gap.build(this->l9_exp.get_output());
        this->l10_cp.build(this->l9_gap.get_output());

        this->l9_exp.call(this->l9_dp.get_output());
        this->l9_dp.get_output().free_element();

        this->l9_gap.call(this->l9_exp.get_output());
        this->l9_exp.get_output().free_element();

        this->l10_cp.call(this->l9_gap.get_output());
        this->l9_gap.get_output().free_element();
`
但是效果还是很差，应该往哪个方向定位原因？

Auroragan, 2021-09-10:
@xiao-mb export 参数目前有些bug, 等会儿会推一版上去

TiramisuJ, 2021-09-10:
@xiao-mb 已推， 再试试

xiao-mb, 2021-09-10:
情况有所改善，之前不同的输入图片预测的是同一值，现在不同的输入图片预测值也在变，但是还是不对，我先检查下camera的图像吧