espressif/esp-dl #62: 执行example.py出现了错误。

xiao-mb, 2021-09-10:
当执行/esp-dl/tools/quantization_tool/examples/example.py出现了如下错误：
(py37) gc@gc-VirtualBox:/media/sf_share/esp-dl/tools/quantization_tool/examples$ python example.py
Traceback (most recent call last):
  File "example.py", line 32, in <module>
    model_proto = onnx.load(optimized_model_path)
  File "/home/gc/anaconda3/envs/py37/lib/python3.7/site-packages/onnx/__init__.py", line 120, in load_model
    s = _load_bytes(f)
  File "/home/gc/anaconda3/envs/py37/lib/python3.7/site-packages/onnx/__init__.py", line 34, in _load_bytes
    with open(cast(Text, f), 'rb') as readable:
TypeError: expected str, bytes or os.PathLike object, not NoneType

xiao-mb, 2021-09-10:
同样的环境下，上一个版本是可以的

Auroragan, 2021-09-10:
example 是没有任何修改的, 这个错是报在 onnx 的, 麻烦确认下你的路径 optimized_model_path 是否正确

xiao-mb, 2021-09-10:
optimized_model_path的路径写死，就可以了，但是上一个版本可以正常运行，不知为什么，难道是 optimized_model_path = optimize_fp_model(model_path) 这个函数修改了

Auroragan, 2021-09-10:
Optimizer.py 并没有做任何修改，我们这边测试也是没问题的

xiao-mb, 2021-09-10:
生成的函数名里为啥有“/"啊

`#pragma once

#include <stdint.h>
#include "dl_constant.hpp"

namespace test_order
{
    const dl::Filter<int16_t> *get_test_model/l1_conv/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l1_conv/Conv2D_bias();
    const dl::Activation<int16_t> *get_test_model/l1_conv/Conv2D_activation();
    const dl::Filter<int16_t> *get_test_model/l2_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l2_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l2_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l2_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l2_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l3_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l3_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l3_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l3_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l3_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l4_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l4_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l4_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l4_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l4_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l5_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l5_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l5_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l5_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l5_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l6_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l6_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l6_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l6_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l6_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l7_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l7_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l7_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l7_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l7_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l8_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l8_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l8_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l8_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l8_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_test_model/l9_dp/depthwise_filter();
    const dl::Bias<int16_t> *get_test_model/l9_dp/depthwise_bias();
    const dl::Activation<int16_t> *get_test_model/l9_dp/depthwise_activation();
    const dl::Filter<int16_t> *get_test_model/l9_exp/Conv2D_filter();
    const dl::Bias<int16_t> *get_test_model/l9_exp/Conv2D_bias();
    const dl::Filter<int16_t> *get_fused_gemm_0_filter();
    const dl::Bias<int16_t> *get_fused_gemm_0_bias();
}
`

Auroragan, 2021-09-10:
名字是直接从你的模型里拿出来的，麻烦先手动替换一下，之前没测到这种命名情况

Auroragan, 2021-09-10:
或者你调一下optimizer.py 里的rename_node

xiao-mb, 2021-09-10:
好的

closed 2021-09-10 by xiao-mb