espressif/esp-dl #64: set_exponent() 这个指数应该如何设置？

xiao-mb, 2021-09-13:
把网络移植到s3上后，识别不正确。评估的准确率如下：

> Evaluating the performance on esp32s3:
>accuracy of int16 model is: 0.996094
>accuracy of fp32 model is: 0.996094
把同张图片导出，在PC上测试，识别正确。在S3上，不同的图片大多数会预测出同一个值，而且不正确。这个问题应该从哪些方面来定位原因？

_Originally posted by @xiao-mb in https://github.com/espressif/esp-dl/issues/58#issuecomment-917624528_

exponent的值原来为-8 改为-15，不同的图像，大部分会有不同的预测值，但是还是不对，猜测s3上的准确率不行可能与exponent这个值有关。所以这个值应该怎么设置？
https://github.com/espressif/esp-dl/blob/dcaaa939edeef575a7340ff28dcfe8ff1411b988/tutorial/main/app_main.cpp#L314

训练模型用的数据集除了256，量化工具用的数据集也除了256，量化工具输出的指数为 -15 ，s3的像素点范围0-255。

yehangyang, 2021-09-13:
> 训练模型用的数据集除了256，量化工具用的数据集也除了256，量化工具输出的指数为 -15 ，s3的像素点范围0-255。

@xiao-mb 如果训练时除了 256， 那么 input 的 exponent 设定为 -8。

PureHing, 2021-09-14:
@yehangyang 这个我也很困惑。如果我采用8-bit量化，训练时数据img=img/255；那么对于esp32s3的预处理应该是如何？比如使用`convert_to_u8.py`得打unit8_t的数据img1，需要执行img1=round(img1/255*2^|exponent|)吗
input.set_element((int8_t *)img1).set_exponent(-8).set_shape({28, 28, 3}).set_auto_free(false);

yehangyang, 2021-09-15:
@PureHing 只要保证 quant*2^exponent = float 。如果训练的时候 img2_f=img1_f/255，img2_f 的取值范围是[0,1]。 在对 img2_f 做量化的时候：
- int16_t 量化： 
    - 1/255 约等于 2^-8
    - 所以 input.set_element(img1_f).set_exponent(-8)....
- int8_t 量化：
    - 注意，int8_t（-128,127） 是放不下 uint8_t（0,255） 的
    - 所以 uint8_t 的 img1_f 需要首先除以 2 转换到 int8_t, imgb_f=img1_f/2
    - 所以 input.set_element(img1_f/2).set_exponent(-7)...

对于上述的 int8_t 量化，其实有 1-bit 的精度损失。所以我建议，如果打算用 int8_t 量化的，
- 训练的时候，用这样的归一化：(img1_f - 128)/128
- 这样 input.set_element(img1_f - 128).set_exponent(-7)...

PureHing, 2021-09-15:
@yehangyang thanks

closed 2021-09-18 by xiao-mb