espressif/esp-dl #65: assert failed: block_merge_prev heap_tlsf.c:350 (block_is_free(prev) && "prev block is not free though marked as such")

PureHing, 2021-09-15:
simple main code:
```
    int8_t *score = model.out_layer.get_output().get_element_ptr();
    if (NULL == score)
    {
        printf("failed\n");
    }
    float *result = (float *)malloc(size_out);
    for(int i=0;i<size_out;i++)
    {
        printf("%d:",i);
        // printf("%d,",score[i]);
        result[i]=score[i]*pow(2.f,-3);
        printf("%f,",result[i]);
    }
    printf("\nsuccess\n");
```
**What is the cause of this problem？**
Log:
```
Size_out:450
0:1:2:3:4:5:6:7:8:9:10:11:12:13:14:15:16:17:18:19:20:21:22:23:24:25:26:27:28:29:30:31:32:33:34:35:36:37:38:39:40:41:42:43:44:45:46:47:48:49:50:51:52:53:54:55:56:57:58:59:60:61:62:63:64:65:66:67:68:69:70:71:72:73:74:75:76:77:78:79:80:81:82:83:84:85:86:87:88:89:90:91:92:93:94:95:96:97:98:99:100:101:102:103:104:105:106:107:108:109:110:111:112:113:114:115:116:117:118:119:120:121:122:123:124:125:126:127:128:129:130:131:132:133:134:135:136:137:138:139:140:141:142:143:144:145:146:147:148:149:150:151:152:153:154:155:156:157:158:159:160:161:162:163:164:165:166:167:168:169:170:171:172:173:174:175:176:177:178:179:180:181:182:183:184:185:186:187:188:189:190:191:192:193:194:195:196:197:198:199:200:201:202:203:204:205:206:207:208:209:210:211:212:213:214:215:216:217:218:219:220:221:222:223:224:225:226:227:228:229:230:231:232:233:234:235:236:237:238:239:240:241:242:243:244:245:246:247:248:249:250:251:252:253:254:255:256:257:258:259:260:261:262:263:264:265:266:267:268:269:270:271:272:273:274:275:276:277:278:279:280:281:282:283:284:285:286:287:288:289:290:291:292:293:294:295:296:297:298:299:300:301:302:303:304:305:306:307:308:309:310:311:312:313:314:315:316:317:318:319:320:321:322:323:324:325:326:327:328:329:330:331:332:333:334:335:336:337:338:339:340:341:342:343:344:345:346:347:348:349:350:351:352:353:354:355:356:357:358:359:360:361:362:363:364:365:366:367:368:369:370:371:372:373:374:375:376:377:378:379:380:381:382:383:384:385:386:387:388:389:390:391:392:393:394:395:396:397:398:399:400:401:402:403:404:405:406:407:408:409:410:411:412:413:414:415:416:417:418:419:420:421:422:423:424:425:426:427:428:429:430:431:432:433:434:435:436:437:438:439:440:441:442:443:444:445:446:447:448:449:
success

assert failed: block_merge_prev heap_tlsf.c:350 (block_is_free(prev) && "prev block is not free though marked as such")

Backtrace:0x40379b02:0x3fcb72d00x40396db9:0x3fcb72f0 0x4039bf66:0x3fcb7310 0x4039b589:0x3fcb7430 0x4039bbe6:0x3fcb7450 0x40379cc9:0x3fcb7470 0x4039bfb5:0x3fcb7490 0x4200be19:0x3fcb74b0 0x42005857:0x3fcb74d0 0x420078fc:0x3fcb74f0 0x4201cc55:0x3fcb78f0 
0x40379b02: panic_abort at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/esp_system/panic.c:402

0x40396db9: esp_system_abort at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/esp_system/esp_system.c:129

0x4039bf66: __assert_func at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/newlib/assert.c:85

0x4039b589: block_merge_prev at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/heap_tlsf.c:350
 (inlined by) tlsf_free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/heap_tlsf.c:868

0x4039bbe6: multi_heap_free_impl at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/multi_heap.c:220
 (inlined by) multi_heap_free_impl at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/multi_heap.c:209

0x40379cc9: heap_caps_free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/heap_caps.c:305

0x4039bfb5: free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/newlib/heap.c:46

0x4200be19: operator delete(void*) at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49

0x42005857: __gnu_cxx::new_allocator<int>::deallocate(int*, unsigned int) at /home/librazxc/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/ext/new_allocator.h:125
 (inlined by) std::allocator_traits<std::allocator<int> >::deallocate(std::allocator<int>&, int*, unsigned int) at /home/librazxc/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/alloc_traits.h:462
 (inlined by) std::_Vector_base<int, std::allocator<int> >::_M_deallocate(int*, unsigned int) at /home/librazxc/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h:304
 (inlined by) std::_Vector_base<int, std::allocator<int> >::~_Vector_base() at /home/librazxc/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h:285
 (inlined by) std::vector<int, std::allocator<int> >::~vector() at /home/librazxc/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h:570
 (inlined by) dl::Tensor<signed char>::~Tensor() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/typedef/dl_variable.hpp:78
 (inlined by) dl::layer::MaxPool2D<signed char>::~MaxPool2D() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/layer/dl_layer_max_pool2d.hpp:67
 (inlined by) DMMODEL::~DMMODEL() at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../model/dm_model.hpp:21

0x420078fc: app_main at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../main/app_main.cpp:82

0x4201cc55: main_task at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/freertos/port/port_common.c:136
```

Auroragan, 2021-09-15:
did you free the output of maxpool layer?

PureHing, 2021-09-15:
```
    void call(Tensor<int8_t> &input)
    {
        //TODO
        this->l0.call(input);
        input.free_element();
        this->l2_concat.calloc_element();
        this->lm_0.call(this->l0.get_output());
        
        this->l1.call(this->l0.get_output());
        this->l0.get_output().free_element();
        this->l2.call(this->l1.get_output());
        this->l1.get_output().free_element();

        this->l3.call(this->l2_concat.get_output());
        this->l2_concat.get_output().free_element();

        this->lm_1.call(this->l3.get_output());
        this->l3.get_output().free_element();

        this->l4.call(this->lm_1.get_output());
        this->lm_1.get_output().free_element();

        this->lm_2.call(this->l4.get_output());
        this->l4.get_output().free_element();

        this->l5.call(this->lm_2.get_output());
        this->lm_2.get_output().free_element();

        this->l6.call(this->l5.get_output());
        this->l5.get_output().free_element();

        this->l7.call(this->l6.get_output());
        this->l6.get_output().free_element();

        this->l8.call(this->l7.get_output());
        this->l7.get_output().free_element();

        this->lm_3.call(this->l8.get_output());
        this->l8.get_output().free_element();

        this->l9.call(this->lm_3.get_output());
        this->lm_3.get_output().free_element();

        this->l10.call(this->l9.get_output());
        this->l9.get_output().free_element();

        this->l11.call(this->l10.get_output());
        this->l10.get_output().free_element();

        this->l12.call(this->l11.get_output());
        this->l11.get_output().free_element();
    }
```
As shown above, one(lm_0) of the maxpool layers is in l2_concat, I released l2_concat layer.
and lm_1 & lm_2 & lm_3 is also maxpooling layer.

yehangyang, 2021-09-15:
> int8_t *score = model.out_layer.get_output().get_element_ptr();

@PureHing, I can not find the out_layer in the call() function.

PureHing, 2021-09-15:
@yehangyang  sorry ,out_layer is l12

yehangyang, 2021-09-15:
@PureHing, the error happens in MaxPool2D's deconstruction. We'll check the source code. Is the score result right?

PureHing, 2021-09-15:
@yehangyang the model is regression nework , now i can't parse the score result data ,it interupt here

> result[i]=score[i]*pow(2.f,-3);

yehangyang, 2021-09-15:
@PureHing, 
I saw your log printed 'success'. 
> printf("%f,",result[i]);
It didn't work?

How about replace with `score[i]*0.125`?

PureHing, 2021-09-16:
@yehangyang After print `success`, then an error occurs, and the subsequent parse code cannot be executed.

yehangyang, 2021-09-16:
> 0x420079ca: app_main at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../main/app_main.cpp:127
@PureHing, What is line 127?

PureHing, 2021-09-16:
@yehangyang 

> dbox *boxes = (dbox *)calloc(box * (1 + 4), sizeof(dbox));

yehangyang, 2021-09-16:
> float *result = (float *)malloc(size_out);

@PureHing should be float *result = (float *)malloc(size_out*sizeof(float));

PureHing, 2021-09-16:
@yehangyang Oh, it's very awkward.

```
success
malloc
decode

resultData.detect_num=6 
i:0 ID:dm: 55%  left:39.497913 right:39.503384 top:0.000000 bottom:30.323515
i:1 ID:dm: 97%  left:37.499355 right:43.499355 top:36.500000 bottom:43.500000
i:2 ID:dm: 52%  left:49.612305 right:57.544346 top:38.986656 bottom:39.023701
i:3 ID:dm: 97%  left:53.994823 right:59.994827 top:52.911259 bottom:59.088741
i:4 ID:dm: 55%  left:71.774185 right:77.069168 top:55.993240 bottom:56.006756
i:5 ID:dm: 92%  left:70.835175 right:76.130157 top:68.911255 bottom:75.088737

assert failed: tlsf_free heap_tlsf.c:866 (!block_is_free(block) && "block already marked as free")

Backtrace:0x40379b02:0x3fcb5ff00x40396db9:0x3fcb6010 0x4039bf66:0x3fcb6030 0x4039b536:0x3fcb6150 0x4039bbe6:0x3fcb6170 0x40379cc9:0x3fcb6190 0x4039bfb5:0x3fcb61b0 0x42005492:0x3fcb61d0 0x42007b11:0x3fcb61f0 0x4201dcd9:0x3fcb78f0 
0x40379b02: panic_abort at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/esp_system/panic.c:402

0x40396db9: esp_system_abort at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/esp_system/esp_system.c:129

0x4039bf66: __assert_func at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/newlib/assert.c:85

0x4039b536: tlsf_free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/heap_tlsf.c:866 (discriminator 1)

0x4039bbe6: multi_heap_free_impl at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/multi_heap.c:220
 (inlined by) multi_heap_free_impl at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/multi_heap.c:209

0x40379cc9: heap_caps_free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/heap/heap_caps.c:305

0x4039bfb5: free at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/newlib/heap.c:46

0x42005492: dl::tool::free_aligned(void*) at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/tool/dl_tool.hpp:140
 (inlined by) dl::Tensor<signed char>::free_element() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/typedef/dl_variable.hpp:382
 (inlined by) dl::Tensor<signed char>::free_element() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/typedef/dl_variable.hpp:378
 (inlined by) dl::Tensor<signed char>::~Tensor() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/typedef/dl_variable.hpp:77
 (inlined by) dl::layer::Conv2D<signed char>::~Conv2D() at /home/librazxc/SoftwareDevelop/esp/esp-dl/include/layer/dl_layer_conv2d.hpp:77
 (inlined by) DMMODEL::~DMMODEL() at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../model/dm_model.hpp:21

0x42007b11: app_main at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../main/app_main.cpp:103

0x4201dcd9: main_task at /home/librazxc/SoftwareDevelop/esp/esp-idf/components/freertos/port/port_common.c:136
```

yehangyang, 2021-09-16:
> (inlined by) DMMODEL::~DMMODEL() at /home/librazxc/SoftwareDevelop/esp/dl_dm_detection_esp32s3/build/../model/dm_model.hpp:21
@PureHing What is this?

PureHing, 2021-09-16:
[image]

PureHing, 2021-09-16:
> i:0 ID:dm: 55%  left:39.497913 right:39.503384 top:0.000000 bottom:30.323515
> i:1 ID:dm: 97%  left:37.499355 right:43.499355 top:36.500000 bottom:43.500000
> i:2 ID:dm: 52%  left:49.612305 [...]

@yehangyang the result is wrong,i will check it later.

yehangyang, 2021-09-16:
@PureHing , What your ```DMMODEL::~DMMODEL()```

PureHing, 2021-09-16:
@yehangyang Call the parent class destructor, I did not define.

yehangyang, 2021-09-16:
@PureHing, is ```l12``` declared in public?

PureHing, 2021-09-16:
```
#pragma once
#include "dl_layer_model.hpp"
#include "dl_layer_conv2d.hpp"
#include "dl_layer_max_pool2d.hpp"
#include "dl_layer_concat2d.hpp"
#include "dm_coefficient.hpp"
#include <stdint.h>

using namespace dl;
using namespace layer;
using namespace dm_coefficient;

class DMMODEL : public Model<int8_t>
{
private:
public:
    //Reshape layer name: model/conv2d/Conv2D__10
    Conv2D<int8_t> l0; //Conv layer name: model_conv2d_Conv2D, output_exponent: -11
    MaxPool2D<int8_t> lm_0;//MaxPool layer name: model/max_pooling2d/MaxPool
    Conv2D<int8_t> l1;//Conv layer name: model/conv2d_1/Conv2D,
    Conv2D<int8_t> l2;
    Concat2D<int8_t> l2_concat;
    Conv2D<int8_t> l3;
    MaxPool2D<int8_t> lm_1;   
    Conv2D<int8_t> l4;
    MaxPool2D<int8_t> lm_2; 
    Conv2D<int8_t> l5;
    Conv2D<int8_t> l6;
    Conv2D<int8_t> l7;
    Conv2D<int8_t> l8;
    MaxPool2D<int8_t> lm_3; 
    Conv2D<int8_t> l9;
    Conv2D<int8_t> l10;
    Conv2D<int8_t> l11; //Conv layer name: model_conv2d_11_Conv2D
    

    Conv2D<int8_t> l12; //Conv layer name: model_conv2d_12_BiasAdd

    DMMODEL() : l0(Conv2D<int8_t>(-4,get_model_conv2d_Conv2D_filter(),get_model_conv2d_Conv2D_bias(),get_model_conv2d_Conv2D_activation(),PADDING_SAME,1,1,"l0")),
                lm_0(MaxPool2D<int8_t>({2,2},PADDING_VALID,2,2,"lm_0")), 
                l1(Conv2D<int8_t>(-4,get_model_conv2d_1_Conv2D_filter(),get_model_conv2d_1_Conv2D_bias(),get_model_conv2d_1_Conv2D_activation(),PADDING_SAME,1,1,"l1")),
                l2(Conv2D<int8_t>(-4,get_model_conv2d_2_Conv2D_filter(),get_model_conv2d_2_Conv2D_bias(),get_model_conv2d_2_Conv2D_activation(),PADDING_SAME,2,2,"l2")),
                l2_concat("l2_concat"),
                l3(Conv2D<int8_t>(-4,get_model_conv2d_3_Conv2D_filter(),get_model_conv2d_3_Conv2D_bias(),get_model_conv2d_3_Conv2D_activation(),PADDING_SAME,1,1,"l3")),
                lm_1(MaxPool2D<int8_t>({2,2},PADDING_VALID,2,2,"lm_1")), 
                l4(Conv2D<int8_t>(-4,get_model_conv2d_4_Conv2D_filter(),get_model_conv2d_4_Conv2D_bias(),get_model_conv2d_4_Conv2D_activation(),PADDING_SAME,1,1,"l4")),
                lm_2(MaxPool2D<int8_t>({2,2},PADDING_VALID,2,2,"lm_2")),
                l5(Conv2D<int8_t>(-4,get_model_conv2d_5_Conv2D_filter(),get_model_conv2d_5_Conv2D_bias(),get_model_conv2d_5_Conv2D_activation(),PADDING_SAME,1,1,"l5")),
                l6(Conv2D<int8_t>(-4,get_model_conv2d_6_Conv2D_filter(),get_model_conv2d_6_Conv2D_bias(),get_model_conv2d_6_Conv2D_activation(),PADDING_VALID,1,1,"l6")),
                l7(Conv2D<int8_t>(-4,get_model_conv2d_7_Conv2D_filter(),get_model_conv2d_7_Conv2D_bias(),get_model_conv2d_7_Conv2D_activation(),PADDING_SAME,1,1,"l7")),
                l8(Conv2D<int8_t>(-4,get_model_conv2d_8_Conv2D_filter(),get_model_conv2d_8_Conv2D_bias(),get_model_conv2d_8_Conv2D_activation(),PADDING_VALID,1,1,"l8")),
                lm_3(MaxPool2D<int8_t>({2,2},PADDING_VALID,2,2,"lm_3")),
                l9(Conv2D<int8_t>(-5,get_model_conv2d_9_Conv2D_filter(),get_model_conv2d_9_Conv2D_bias(),get_model_conv2d_9_Conv2D_activation(),PADDING_SAME,1,1,"l9")),
                l10(Conv2D<int8_t>(-4,get_model_conv2d_10_Conv2D_filter(),get_model_conv2d_10_Conv2D_bias(),get_model_conv2d_10_Conv2D_activation(),PADDING_SAME,1,1,"l10")),
                l11(Conv2D<int8_t>(-5,get_model_conv2d_11_Conv2D_filter(),get_model_conv2d_11_Conv2D_bias(),get_model_conv2d_11_Conv2D_activation(),PADDING_VALID,1,1,"l11")),
                l12(Conv2D<int8_t>(-2,get_model_conv2d_12_BiasAdd_filter(),get_model_conv2d_12_BiasAdd_bias(),NULL,PADDING_VALID,1,1,"l12")) {}
                
    void build(Tensor<int8_t> &input)
    {
        //TODO
        this->l0.build(input);
        this->lm_0.build(this->l0.get_output());
        this->l1.build(this->l0.get_output());
        this->l2.build(this->l1.get_output());
        this->l2_concat.build({&this->lm_0.get_output(),&this->l2.get_output()});
        this->l3.build(this->l2_concat.get_output());
        this->lm_1.build(this->l3.get_output());
        this->l4.build(this->lm_1.get_output());
        this->lm_2.build(this->l4.get_output());
[... 41 lines cut ...]
        this->l5.get_output().free_element();

        this->l7.call(this->l6.get_output());
        this->l6.get_output().free_element();

        this->l8.call(this->l7.get_output());
        this->l7.get_output().free_element();

        this->lm_3.call(this->l8.get_output());
        this->l8.get_output().free_element();

        this->l9.call(this->lm_3.get_output());
        this->lm_3.get_output().free_element();

        this->l10.call(this->l9.get_output());
        this->l9.get_output().free_element();

        this->l11.call(this->l10.get_output());
        this->l10.get_output().free_element();

        this->l12.call(this->l11.get_output());
        this->l11.get_output().free_element();
    }
};
```

closed 2021-09-16 by PureHing