espressif/esp-dl #66: model forward result is wrong

PureHing, 2021-09-16:
quantization_tool.py
```
if __name__ == "__main__":
    # Optimize the onnx model
    model_path = 'yolo80.onnx'
    optimized_model_path = optimize_fp_model(model_path)

    # Calibration
    with open('test80.pickle', 'rb') as f:
        test_images = pickle.load(f)

    # Prepare the calibration dataset
    calib_dataset = np.expand_dims(test_images,axis=-1)
    pickle_file_path = 'dm_calib_80.pickle'
    model_proto = onnx.load(optimized_model_path)

    print('Generating the quantization table:')
    calib = Calibrator('int16', 'per-tensor', 'minmax')##Calibrator('int8', 'per-tensor', 'entropy')
    calib.set_providers(['CPUExecutionProvider'])
    calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)
    calib.export_coefficient_to_cpp(model_proto, pickle_file_path, 'esp32s3', '.', 'dm_coefficient', True)
    # shutil.move("./dm_coefficient.cpp","../model/dm_coefficient.cpp")
    # shutil.move("./dm_coefficient.hpp","../model/dm_coefficient.hpp")
    # Evaluate the performance
    print('Evaluating the performance on esp32s3:')
    eva = Evaluator('int16', 'esp32s3')
    eva.set_providers(['CPUExecutionProvider'])
    eva.generate_quantized_model(model_proto, pickle_file_path)

    output_names = [n.name for n in model_proto.graph.output]
    providers = ['CPUExecutionProvider']
    m = rt.InferenceSession(optimized_model_path, providers=providers)

    batch_size = 1
    batch_num = int(len(test_images) / batch_size)
    res = 0
    fp_res = 0
    input_name = m.get_inputs()[0].name

    for i in range(1):
        # int16_model
        [outputs, _] = eva.evalute_quantized_model(calib_dataset[i * batch_size:(i + 1) * batch_size], False)
        check("int16 reslut:",outputs)
        # floating-point model
        fp_outputs = m.run(output_names, {input_name: calib_dataset[i * batch_size:(i + 1) * batch_size].astype(np.float32)})
        # fp_res = fp_res + sum(np.argmax(fp_outputs[0], axis=1) == test_labels[i * batch_size:(i + 1) * batch_size])
        check("fp32 result:",fp_outputs)

        print("end")
```

Log:
```
int16 []
int16
fp32 [array([57.68230438, 43.56007004, 63.74114227, 49.40076065,  0.99928188,
        0.        ]), array([30.37335968, 56.07011414, 36.052742  , 61.44461823,  0.95321083,
        0.        ])]
fp32
end
```

@Auroragan @yehangyang

Sorry to disturb you, the simulation result is wrong, where can I locate the problem.

yehangyang, 2021-09-17:
@PureHing I'm sorry to say you have to check the result layer by layer. Otherwise, we can not locate the problem.
You can 
1. export the fp32 layer result by (* 2 ^ -exponent)
2. save it as a array in C/C++
3. check the result correctness by [Tensor.check_element()](https://github.com/espressif/esp-dl/blob/dcaaa939edeef575a7340ff28dcfe8ff1411b988/include/typedef/dl_variable.hpp#L647)

PureHing, 2021-09-18:
@yehangyang 
Hi,It seems that my model output has a problem from the first layer.
first layer(name: l0): conv2d,1x80x80x1->1x80x80x4, kernel size:3x3,pad=(1,1,1,1,)

My steps:
   1.get fp32 result of first conv2d layer from onnx model
   2.on eps32s3:
```
    int16_t *score = model.l0.get_output().get_element_ptr();
    int size_out = model.l0.get_output().get_size();
    for (int i = 0; i < size_out; i++)
    {
        result[i] = score[i] * pow(2.f, -12);
        printf("%f,",result[i]);
    }
```
the `size_out` is 82*82*4
3.The result gap is very big.

yehangyang, 2021-09-18:
@PureHing 
- export result from PC as ground truth
- the result from PC should be in fp32
- so we turn it to quantization, by ( * 2 ^ - exponent)
- save the ground truth quantization result as C/C++ array
- check the result correctness by [Tensor.check_element()](https://github.com/espressif/esp-dl/blob/dcaaa939edeef575a7340ff28dcfe8ff1411b988/include/typedef/dl_variable.hpp#L647)

PureHing, 2021-09-18:
@yehangyang Hi
code:
```
    input.set_element(IMAGE_ELEMENT).set_exponent(-8).set_shape({IMAGE_HEIGHT, IMAGE_WIDTH, IMAGE_CHANNEL}).set_auto_free(false);
    input.print_shape();
    DMMODEL model;
    // model forward
    latency.start();
    model.forward(input);
    latency.end();
    latency.print("detect", "forward");
    // parse
    printf("\n--------\n");
    int size_out = model.l0.get_output().get_size();
    model.l0.get_output().print_shape();
    printf("\nSize_out:%d\n", size_out);
    int16_t *score = model.l0.get_output().get_element_ptr();
    if (NULL == score)
    {
        printf("failed\n");
    }
    model.l0.get_output().check_element((int16_t *)l0data);
    printf("\nend\n");
```

Log:
```
shape = (0 + 80 + 0, 0 + 80 + 0, 1(1))
detect::forward: 20965 us

--------
shape = (1 + 80 + 1, 1 + 80 + 1, 4(4))

Size_out:26896
shape = (1 + 80 + 1, 1 + 80 + 1, 4(4))
element[0, 0, 0]: 32767 v.s. -8918

end

```

the firtst element is wrong . the same my way.

yehangyang, 2021-09-18:
@PureHing Would you mind changing the **first** layer's padding style from "SAME" to "VALID". There is some problem with it. **We've planed to cancel the "padding in advance" strategy in the next version**. At that time, this problem will not happen.

PureHing, 2021-09-18:
@yehangyang Does it  only need the first layer need to be designed like this`l1(Conv2D<int16_t>(-2, get_l1_filter(), get_l1_bias(), get_l1_activation(), PADDING_VALID, 2, 2, "l1"))`?

yehangyang, 2021-09-18:
@PureHing , Yes, just avoid the first layer using PADDING_SAME.

PureHing, 2021-09-18:
let me try.

PureHing, 2021-09-18:
@yehangyang stride=2 or maxpooling downsample follow floor or ceil?

yehangyang, 2021-09-18:
@PureHing, follow, it should be same as Tensorflow's

PureHing, 2021-09-18:
@yehangyang ok  thanks

PureHing, 2021-09-18:
> input.set_padding_size({1,1,1,1})

```

    input.set_element(IMAGE_ELEMENT).set_exponent(-8).set_shape({IMAGE_HEIGHT, IMAGE_WIDTH, IMAGE_CHANNEL}).set_auto_free(false);
    input.set_padding_size({1,1,1,1});
    model.forward(input);
```

is it right?

yehangyang, 2021-09-18:
@PureHing , Yes.

PureHing, 2021-09-18:
> > input.set_padding_size({1,1,1,1})
> 
> ```
> 
>     input.set_element(IMAGE_ELEMENT).set_exponent(-8).set_shape({IMAGE_HEIGHT, IMAGE_WIDTH, IMAGE_CHANNEL}).set_auto_free(false);
>     input.set_padding_size({1,1,1,1});
>     model.forward(input);
> ```
> 
> is it right?
@yehangyang 
the result is wrong by this way.

PureHing, 2021-09-18:
> Or, You can call input.set_padding_size({1,1,1,1}), just before send input into Model.call(input), right now.

the inference is wrong by this way.

Auroragan, 2021-09-18:
@PureHing Didn't you say the result is correct on the call before?

PureHing, 2021-09-18:
@Auroragan

> Would you mind changing the **first** layer's padding style from "SAME" to "VALID".

I changed my model to avoid the padding problem in firtst layer,then forward output is correct between on esp32s3 comparing to PC.
If I use the old model which first layer padding is "SAME" and adding `input.set_padding_size({1,1,1,1})`, then forward output is wrong on esp32s3 comparing to PC.

xiao-mb, 2021-09-18:
> @yehangyang
> Hi,It seems that my model output has a problem from the first layer.
> first layer(name: l0): conv2d,1x80x80x1->1x80x80x4, kernel size:3x3,pad=(1,1,1,1,)
> 
> My steps:
> 1.get fp32 result of first conv2d layer from onnx model
> 2.on eps32s3:
> 
> ```
>     int16_t *score = model.l0.get_output().get_element_ptr();
>     int size_out = model.l0.get_output().get_size();
>     for (int i = 0; i < size_out; i++)
>     {
>         result[i] = score[i] * pow(2.f, -12);
>         printf("%f,",result[i]);
>     }
> ```
> 
> the `size_out` is 26896 , //80x80x4=25600 with padding to 26896
> 3.The result gap is very big.
> 4.
> 
> > * export the fp32 layer result by (* 2 ^ -exponent)
> > * save it as a array in C/C++
> 
> Do you mean the result on esp32s3?

@PureHing Hi, can you share your code of step 1?

PureHing, 2021-09-20:
> If I use the old model which first layer padding is "SAME" and adding `input.set_padding_size({1,1,1,1})`, then forward output is wrong on esp32s3 comparing to PC.

@yehangyang HI
Am I using this method wrong?

PureHing, 2021-09-22:
@yehangyang @Auroragan The first layer is such a configuration, the result is still incorrect.
model like this:
[image]

Auroragan, 2021-09-22:
@PureHing the result of first layer or the final result?

PureHing, 2021-09-22:
@Auroragan
Neither is right

Auroragan, 2021-09-22:
> > Would you mind changing the first layer's padding style from "SAME" to "VALID".
> 
> I changed my model to avoid the padding problem in firtst layer,then forward output is correct on esp32s3 comparing to PC.

the one you just showed is another model using PADDING_VALID without adding set_padding_size?

PureHing, 2021-09-22:
input setting like this
`input.set_element(IMAGE_ELEMENT).set_exponent(-8).set_shape({IMAGE_HEIGHT,IMAGE_WIDTH, IMAGE_CHANNEL}).set_auto_free(false);`

Auroragan, 2021-09-22:
@PureHing Is there any difference between this model that gets the wrong result and the previous model that's correct?

PureHing, 2021-09-22:
no

Auroragan, 2021-09-22:
yes

PureHing, 2021-09-22:
@Auroragan @yehangyang
Have you received my model and engineering code？I sent it to shiqiqi.

Auroragan, 2021-09-22:
yeah got it. Can you also provide the pickle file and the test input for pc (npy file)?

PureHing, 2021-09-22:
@Auroragan Please give me your email address

Auroragan, 2021-09-22:
please send to shiqiqi

PureHing, 2021-09-22:
Got it

PureHing, 2021-09-22:
@Auroragan 
```
    with open('test80.pickle', 'rb') as f:
        test_images = pickle.load(f)
    calib_dataset = np.expand_dims(test_images,axis=1)
```

```
Conv layer name: Conv_0, output_exponent: -12
Conv layer name: Conv_2, output_exponent: -12
Conv layer name: Conv_4, output_exponent: -11
Conv layer name: Conv_6, output_exponent: -11
Conv layer name: Conv_8, output_exponent: -11
Conv layer name: Conv_10, output_exponent: -11
Conv layer name: Conv_12, output_exponent: -11
Conv layer name: Conv_14, output_exponent: -11
Conv layer name: Conv_16, output_exponent: -11
Conv layer name: Conv_18, output_exponent: -11
Conv layer name: Conv_20, output_exponent: -12
Conv layer name: Conv_22, output_exponent: -11
Conv layer name: Conv_24, output_exponent: -11
Conv layer name: Conv_26, output_exponent: -11
Conv layer name: Conv_28, output_exponent: -11
Conv layer name: Conv_30, output_exponent: -11
Conv layer name: Conv_32, output_exponent: -12
Conv layer name: Conv_34, output_exponent: -12
Conv layer name: Conv_36, output_exponent: -12
Conv layer name: Conv_38, output_exponent: -12
Conv layer name: Conv_40, output_exponent: -11
```

Auroragan, 2021-09-22:
sorry, I mean the generated pickle file of quantization table. and the floating point input you used as ground truth so that I can know the correct result of each layer.

PureHing, 2021-09-22:
@Auroragan the first layer and last layer gt output is in image.hpp

PureHing, 2021-09-23:
> @Auroragan the gt output of first and last layers are in image.hpp

Does this work for you？

PureHing, 2021-09-23:
> the gt output of first and last layers are in image.hpp

@Auroragan Does this work for you？

Auroragan, 2021-09-23:
hmmm the last layer is not the same as the gt you provided, and Conv_0_data is not the result after relu right? If you can provide the pickle file(quantization table) and the fp input, it'll be easier for me to know what's wrong.

PureHing, 2021-09-23:
[pickle-file.zip](https://github.com/espressif/esp-dl/files/7214714/pickle-file.zip)
@Auroragan

Auroragan, 2021-09-23:
@PureHing problem found:
1. at the beginning, int16 API for S3 only support channel = 16*n, now it doesn't have this limitation but the code for exporting the coefficient for 16-bit is out of date so the filter_element for first layer and last layer is incorrect. will fix it ASAP.
2. For Conv_6, Conv_12, Conv_18, should use PADDING_SAME_MXNET not PADDING_SAME, because different platform has different strategy for padding when stride  = 2, I'm guessing this model is not from tensorflow.
3. the output exponent for Conv_40 should be -11 according to the pickle file you provided

I'm getting the correct result after doing the above modification now.

PureHing, 2021-09-23:
@Auroragan 

Can u send  your modified project to me?

Auroragan, 2021-09-23:
ok

PureHing, 2021-09-23:
@Auroragan 
1.
> 1. at the beginning, int16 API for S3 only support channel = 16*n, now it doesn't have this limitation but the code for exporting the coefficient for 16-bit is out of date so the filter_element for first layer and last layer is incorrect. will fix it ASAP.

For my test model, my channels are all multiples of 16, why the first layer of quantization is wrong.
**If I retrain the model, how can I avoid this error？**

2.
> 2\. For Conv_6, Conv_12, Conv_18, should use PADDING_SAME_MXNET not PADDING_SAME, because different platform has different strategy for padding when stride  = 2, I'm guessing this model is not from tensorflow.

Is the method of s3 downsampling the same as the padding method of maxpooling in tensorflow?

Auroragan, 2021-09-23:
1. the input channel of first layer is 1, please wait for fix. if you want to avoid it for now:
-  for 16-bit. input/output channel should be 8*n, n>=1 (typo before) , first layer use c==3 not c==1
- or use 8-bit
2. what do you mean by s3 downsampling?
3. the out_data is correct, but the initialization of this layer is incorrect:
`Conv_40(Conv2D<int16_t>(-10, get_Conv_40_filter(), get_Conv_40_bias(), NULL, PADDING_VALID, 1, 1, "Conv_40")) {}`
4. it's not a must, if you want to use PADDING_SAME, you need to add input.set_padding_size()
5. it's normal to have some offset due to quantization. the result won't be exactly the same as fp32 result.

PureHing, 2021-09-23:
> 1. the input channel of first layer is 1, please wait for fix.

@Auroragan BTW,When will it be fixed?

the outout channel should be 8*n?

Auroragan, 2021-09-23:
@PureHing will send you the lib first, you can use it without any limitation

PureHing, 2021-09-23:
@Auroragan  sorry to disturb you.
When I replaced the `calibrator.so` which you gave, the requantized `conv_0 filter element` was different from the one you gave me this afternoon. I can parse the correct result that you gave me this afternoon,but incorrect for new lib.

generated by new lib:
```
    const static __attribute__((aligned(16))) int16_t Conv_0_filter_element[] = {
          2077,  -4268,   1845,   7087,   7273,   3863,  -2932,   5966,  -3911,  -5610,  -8973,   7078,   2121,  -7769,   4165,  -4981,  -3858,   1996,   3599,  -5371,   3071,   3103,  -7424,  17486,  -5771,  14175, -20329,   1741,   6661,   2254,  -2040,   3083,   1824,   3516,   1443,   1566,   3942,   5003,  -3172,   -707,  13029,   2335, -10934,    963,   5642,  10067,    489,   -326,  -1002, -19053,    751,   7626,  -9285,  11625,  -6636,   2608,   4242,  -2172,  14463,   -901,   8887,   6514, -10002,    994,    715,   -586,    512,   3579, -21433,  -2547,  22241,  -2228,  -6197, -12812,   1593,   7742,  24893,  -3850,   -209, -11986,    989,   9886,  -7551,   7236,  10315,   1597, -15835,   2864, -11546,   3574,  11251, -14135,  -6187,  13756,     20,  -5940,   5874,   1973,  -4873,   -119,   2842,  -3484,   -273,  -4700,  -3700,   -896,  -5572,  -2543,   5573,   1937,   4858,  -4018,  -3804,   1043,  -6315,  -8327,  -7253, -13258,   7717,   2014,  19076, -14446,  -3168,   4038,  -2180,   1960,   8047,  17826,   2918,   6284,   1141, -11618,  -1900, -11310, -10216,  -5555,    447,   2940,  -8804, -12618,   1900,  16678,   5533, -10553};

```

Auroragan, 2021-09-24:
Have you tried the new one?

> But I need to re-convert the new model. This dynamic library doesn't work for me.

what do you mean?

PureHing, 2021-09-24:
@Auroragan I did not receive a new one today.

I retrained a new model, now I want to convert it and deploy on esp32s3,but the result is incorrect,too.

PureHing, 2021-09-24:
@Auroragan Sorry, can you send it to my email(zhengxc@topband.com.cn), I can’t contact shiqiqi.

PureHing, 2021-09-24:
@Auroragan HI,
Do you use the new dynamic library to test the first layer output of the model I gave you?

> @Auroragan sorry to disturb you.
> When I replaced the `calibrator.so` which you gave, the 16-bit quantized `conv_0 filter element` was different from the one you gave me this afternoon. I can prase the correct result that you gave me this afternoon.
> generated by new lib（Same model） :
> 
> ```
>     const static __attribute__((aligned(16))) int16_t Conv_0_filter_element[] = {
>           2077,  -4268,   1845,   7087,   7273,   3863,  -2932,   5966,  -3911,  -5610,  -8973,   7078,   2121,  -7769,   4165,  -4981,  -3858,   1996,   3599,  -5371,   3071,   3103,  -7424,  17486,  -5771,  14175, -20329,   1741,   6661,   2254,  -2040,   3083,   1824,   3516,   1443,   1566,   3942,   5003,  -3172,   -707,  13029,   2335, -10934,    963,   5642,  10067,    489,   -326,  -1002, -19053,    751,   7626,  -9285,  11625,  -6636,   2608,   4242,  -2172,  14463,   -901,   8887,   6514, -10002,    994,    715,   -586,    512,   3579, -21433,  -2547,  22241,  -2228,  -6197, -12812,   1593,   7742,  24893,  -3850,   -209, -11986,    989,   9886,  -7551,   7236,  10315,   1597, -15835,   2864, -11546,   3574,  11251, -14135,  -6187,  13756,     20,  -5940,   5874,   1973,  -4873,   -119,   2842,  -3484,   -273,  -4700,  -3700,   -896,  -5572,  -2543,   5573,   1937,   4858,  -4018,  -3804,   1043,  -6315,  -8327,  -7253, -13258,   7717,   2014,  19076, -14446,  -3168,   4038,  -2180,   1960,   8047,  17826,   2918,   6284,   1141, -11618,  -1900, -11310, -10216,  -5555,    447,   2940,  -8804, -12618,   1900,  16678,   5533, -10553};
> ```
> 
> But I need to re-convert the new model. This dynamic library doesn't work for me.

Today’s library has the same result as yesterday’s library, which is inconsistent with your hand-tuned

PureHing, 2021-09-25:
The results can be run normally, thank you very much

closed 2021-10-08 by PureHing