espressif/esp-dl #73: 如何将该库移植到已有工程中？

strongerux, 2021-10-30:


yehangyang, 2021-11-02:
可将 esp-dl 作为已有项目的组件。可参考 [esp-who](https://github.com/espressif/esp-who)。

strongerux, 2021-11-02:
比如我需要将哪些文件放在esp-idf的哪个位置？是否有这样的引导？因为似乎环境变量的包含是通过export.bat实现的，应该需要把库文件放在指定路径下才能识别吧。

yehangyang, 2021-11-02:
esp-dl 不是一定要放在 esp-idf 中的。我们已经为 esp-dl 配置好作为组件的 CMake。参考 [esp-who](https://github.com/espressif/esp-who)，将 esp-dl 加入 component，并在具体项目的 CMake 里进行配置即可，如[这样](https://github.com/espressif/esp-who/blob/013561615c99f78ac7fad216e4d66daa146ac409/examples/human_face_detection/lcd/CMakeLists.txt#L5)。

strongerux, 2021-11-02:
如果把它整个放在esp-idf/component目录下也可以吗？

yehangyang, 2021-11-02:
也可以

strongerux, 2021-11-02:
现在出现了一个新问题，我的模型输入是一维的传感器数据，这将导致转换失败，是否还没有适配一维数据？

strongerux, 2021-11-02:
Traceback (most recent call last):
  File "A:/AiThinker/workspace/esp-dl/tools/quantization_tool/step2.py", line 13, in <module>
    optimizer.optimize_fp_model('model.onnx')
  File "A:\AiThinker\workspace\esp-dl\tools\quantization_tool\optimizer.py", line 149, in optimize_fp_model
    model_proto = convert_model_batch_to_dynamic(model_proto)
  File "A:\AiThinker\workspace\esp-dl\tools\quantization_tool\optimizer.py", line 139, in convert_model_batch_to_dynamic
    model = shape_inference.infer_shapes(model)
  File "A:\python3.7\lib\site-packages\onnx\shape_inference.py", line 37, in infer_shapes
    inferred_model_str = C.infer_shapes(model_str, check_type, strict_mode)
RuntimeError: input 1 is out of bounds
报错信息以上。

yehangyang, 2021-11-02:
目前还不支持一维数据，不过你可以尝试用二维的接口来实现一维的操作。

strongerux, 2021-11-02:
好的，最后一个问题，example中的模块读取不到
[image]

yehangyang, 2021-11-02:
请你在什么系统下运行的，目前仅支持 windows 和 linux，python 版本 3.7

strongerux, 2021-11-02:
python3.7.3 windows10

yehangyang, 2021-11-03:
有尝试运行吗？

strongerux, 2021-11-03:
运行提示找不到模块

Auroragan, 2021-11-03:
是在 examples 目录下运行的吗？ conda 环境吗？

Auroragan, 2021-11-03:
> RuntimeError: input 1 is out of bounds

关于这个错，方便提供一下模型吗？

strongerux, 2021-11-03:
没装conda
模型以下
[model.zip](https://github.com/espressif/esp-dl/files/7465339/model.zip)

Auroragan, 2021-11-03:
建议用 conda 试试

https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/optimizer.py#L135 改成这样就不会报错了
```
# model.opset_import[0].version = 13

# if not dim[0].dim_param:
dim[0].dim_param = 'N'
```

但是这个模型看起来很冗余，你的原始模型是什么格式的？

strongerux, 2021-11-03:
好像是.h5转的，好久以前整的了，K210上能用，准备在ESP32上做点工作所以拿来试试。

Auroragan, 2021-11-03:
目前看来这个图需要手动修改才能用 quantization_tool 进行量化，像很多squeeze, unsqueeze 节点其实都没有必要
或者用别的方法确定量化因子后用 convert_tool 直接转参数，可能还方便一些

strongerux, 2021-11-03:
convert_tool这里提到.npy文件，这个文件是哪来的啊？

Auroragan, 2021-11-03:
自行将模型每层的参数保存成相应 npy 文件
可以阅读一下 https://github.com/espressif/esp-dl/blob/master/tools/convert_tool/specification_of_config_json_cn.md

strongerux, 2021-11-03:
了解了，感觉属实是有些繁琐了，TFLM现在还能用吗？

Auroragan, 2021-11-03:
TFLM 目前不是我们在维护

closed 2021-11-16 by strongerux