espressif/esp-dl #74: 模型转换

kazyun, 2021-11-17:
如何将训练好的原生tensorflow或者pytorch等模型转换成你们平台对应的类型呢？ 我看docs中的相关说明，感觉都是基于已经转化后的。

yehangyang, 2021-11-18:
[这个文档](https://github.com/espressif/esp-dl/tree/master/tutorial)讲述了如何基于 ESP-DL 构建自己的模型。
其中 Step1~Step3 是我们保留的一个通用方法，需要自行将参数导出成 .npy，再调用 ESP-DL 提供的脚本转换成 ESP-DL 可接受的 C/C++ 形式。
而我们正在完善这个转化过程，帮助用户去除“导出 .npy”这个过程。请参考 [Quantization Toolkit](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/README.md)，其中的 “Calibrator” 可以帮助你将模型参数直接导出 ESP-DL 可接受的 C/C++ 形式。

而[文档](https://github.com/espressif/esp-dl/tree/master/tutorial)中，剩余的步骤仍需用户手动完成。

yehangyang, 2021-11-18:
首先参考 [Quantization Tool](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/README.md) 的 Optimizer 和 Calibrator 将 Tensorflow 模型中的参数导出。
再参考 [How to Customize a Model Step by Step](https://github.com/espressif/esp-dl/tree/master/tutorial) 中的 Step4~Step5 编写自己的模型。

MichaelDu9226, 2021-12-16:
请问将tensorflow模型转化为DL模型的方法是否有更新？例如Step4~Step5 仍然需要用户自己编写吗？
另外能否出一个更基础更详细的教程，将一下如果将模型转化为DL

目前我们已经可以基于tensorflow lite训练的模型完成分类等应用， 但因为速度太慢，几乎无实用价值，期待DL的加速，期待你们的更新。

> 首先参考 [Quantization Tool](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/README.md) 的 Optimizer 和 Calibrator 将 Tensorflow 模型中的参数导出。你会得到一个源文件和一个头文件，包含了导出的参数。 再参考 [How to Customize a Model Step by Step](https://github.com/espressif/esp-dl/tree/master/tutorial) 中的 Step4~Step5 编写自己的模型。

yehangyang, 2021-12-30:
@MichaelDu9226 目前仍只支持如下方式。Step4~Step5 需要自行编写。
> 首先参考 [Quantization Tool](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/README.md) 的 Optimizer 和 Calibrator 将 Tensorflow 模型中的参数导出。你会得到一个源文件和一个头文件，包含了导出的参数。
> 再参考 [How to [...]

对于教程如有不明白的地方，请指出。大致流程是：
用户自行使用 Tensorflow，Pytorch 或者 MXNet 训练模型 -> 将自己的模型转成 onnx 格式 -> 依次按照上述引用的步骤操作

closed 2022-09-21 by Auroragan