espressif/esp-dl #76: Handpose Recognition

AKEng1, 2021-11-21:
Hi, 

Handpose Recognition is listed as one of the modules in esp-dl. Is there a time line for its availability.

TiramisuJ, 2021-11-22:
Hi
Handpose recognition will be available before the end of this year，You can use this [older version](https://github.com/espressif/esp-dl/tree/420fc7e219ba98e40a5493b9d4be270db2f2d724/pose_estimation) before that.

ElliottSeer, 2022-03-07:
啥时候支持手势识别呀👻

Criminal-9527, 2024-09-26:
@TiramisuJ 
旧版本的这个手势识别好像缺少了一些定义诶，我在链接的时候找不到好多结构体与函数的定义：
undefined reference to `dl_matrix3dqq_blazeblock
undefined reference to `dl_matrix3dqq_double_blazeblock
等等好多，我已经链接了一部分文件进去，但是没法链接这个lib/mkmodel_windows，
```
idf_component_register(
    SRCS
        "main.c"
        "pe_forward.c"
        "image_util.c"
    INCLUDE_DIRS
        "."
        "../include"
)

target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libdetection.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libdl.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libfd.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libfr.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libod.a")
target_link_libraries(${COMPONENT_LIB} INTERFACE "D:/Espressif/frameworks/esp-idf-v5.2.2/examples/dl-camera/lib/libpe.a")
```
如果像上面一样链接mkmodel_windows文件就会报错说无法识别这个文件格式，有什么办法解决吗。
感谢！