espressif/esp-dl #77: Esp-dl VS Tensorflow Lite

dankal444, 2021-11-26:
Hi, 
I am currently using Tensorflow Lite models on ESP32 (https://www.tensorflow.org/lite/microcontrollers). 
I am using mostly conv2d, batchnorm, depth-wise conv2d and relu layers.

Is it worth (and easily achievable) to convert Tensorflow Lite models to esp-dl format. Will there be any performance gain?

TiramisuJ, 2021-11-30:
Hi,
We recommend that you use the esp-dl if you want to get the high performance, there are many optimizations for the basic operators in esp-dl. You can refer to this [tutorial](https://github.com/espressif/esp-dl/tree/master/tutorial) to convert your model.

dankal444, 2021-12-23:
Thank you TiramisuJ for response.

I have an update. Me and my colegues managed to convert 12 layer model with some random weights to esp-dl and (if we did it correctly as it seems) it is ~10x faster than "vanilla" Tensorflow Lite.

But, we are having big problems in achieving compatibility of results between them. Whole process of conversion is complicated and there are many places we might make some error. Now we struggle to convert even only 2-layer model and have the same results. There is some magic about `scale` parameter that we do not understand, during quantization it gets generated automatically, with/without sample database. 

Would it be possible in future to have a tool that just takes tflite model, and outputs all those 3 .cpp files (one could call them layers.cpp, layers.hpp and model.cpp) that esp-dl uses?
We have possibility to use:
-tflite float32 model
-tflite fully converted to int8 model -> this should be fairly easy to convert to esp-dl int8 model right?

ESP-YHY, 2022-01-17:
Hi @dankal444, Thank you for your good advice. So far, we provide a [quantization tool kit](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/README.md) to help produce the `exponent` parameter. It can help convert from an ONNX model to ESP-DL format coefficients. We have a plan to produce a model too in the future.

closed 2024-10-21 by sun-xiangyu