espressif/esp-dl #78: Transpose 层编译失败

notplus, 2021-12-06:
Hi, Transpose layer 的构造函数似乎有问题，环境是 Ubuntu 20.04 g++ 9.3.0，以下为报错信息：

```
/esp32/esp-who/components/esp-dl/include/layer/dl_layer_transpose.hpp: In instantiation of 'dl::layer::Transpose<feature_t>::Transpose(std::vector<int>, const char*, bool) [with feature_t = signed char]':
../model/lpdnet_model.hpp:174:60:   required from here
/esp32/esp-who/components/esp-dl/include/layer/dl_layer_transpose.hpp:25:30: error: 'dl::layer::Transpose<signed char>::perm' will be initialized after [-Werror=reorder]
             std::vector<int> perm;         /*<! the new arangement of the dims. if perm == {}, the dims arangement will be reversed. >*/
                              ^~~~
/esp32/esp-who/components/esp-dl/include/layer/dl_layer_transpose.hpp:23:18: error:   'bool dl::layer::Transpose<signed char>::inplace' [-Werror=reorder]
             bool inplace;                  /*<! true: the output will store to input0
                  ^~~~~~~
In file included from ../model/lpdnet_model.hpp:19,
                 from ../main/app_main.cpp:9:
/esp32/esp-who/components/esp-dl/include/layer/dl_layer_transpose.hpp:36:13: error:   when initialized here [-Werror=reorder]
             Transpose(std::vector<int> perm = {}, const char *name = "Transpose", bool inplace = false) : Layer(name), perm(perm), inplace(inplace), output_shape({})
             ^~~~~~~~~
```

将构造函数修改为如下部分可以正常编译
```
Transpose(std::vector<int> perm = {}, const char *name = "Transpose", bool inplace = false) : Layer(name), inplace(inplace), output_shape({})
{
    this->perm.assign(perm.begin(), perm.end());
}
```

TiramisuJ, 2021-12-07:
Hi
感谢你提出的问题， 这个问题已经修复被修复了，你可以将本地的esp-dl分支更新至最新的master，再次编译试试。

notplus, 2021-12-07:
ok，代码似乎没有 push 到 github，最新的 commit 14小时前

TiramisuJ, 2021-12-07:
的确是14小时前推的，你已经是最新的commit了吗？

notplus, 2021-12-07:
ok，已解决

closed 2021-12-07 by notplus