espressif/esp-dl #81: int8 per-channel entropy convert failed

notplus, 2021-12-07:
Hi, 我在使用量化工具时，`calib = Calibrator('int8', 'per-tensor', 'minmax')` 可以正常使用，但推理结果不正确，使用 `calib = Calibrator('int8', 'per-channel', 'entropy')` 则出现如下错误：
```
Exception has occurred: ValueError
All entries are 0 for this distribution
  File "/esp32/esp-who/components/esp-dl/tools/quantization_tool/examples/convert.py", line 93, in <module>
    calib.generate_quantization_table(model_proto, test_images, pickle_file_path)
```
模型使用了 `Conv2D` `DepthwiseConv2D` `Pad` `Add2D` `Concat` `Transpose` 算子

Auroragan, 2021-12-07:
请问你的原始模型是 onnx 吗？input 是怎么处理的呢？

notplus, 2021-12-07:
原始模型是 TensorFlow ， 转到 onnx ，input 模型训练时 `(img - 128) / 128`，在使用量化工具时，`test_images` 做了同样操作

Auroragan, 2021-12-07:
方便贴一下你的代码和模型吗？

notplus, 2021-12-07:
好的，[代码和模型](https://github.com/espressif/esp-dl/files/7665238/convert.zip)

Auroragan, 2021-12-07:
你说的推理结果是板子上的吗？ 代码中没有看到

notplus, 2021-12-07:
不好意思，板子上的推理结果应该是没问题的，现在就是没办法使用 int8 per-channel entropy

Auroragan, 2021-12-07:
有些分布上的问题，会尽快 fix
我看你用的是 esp32, 如果没有内存上的限制你可以先用 16-bit 量化，32 上 16-bit 推理时间会更短

notplus, 2021-12-07:
好的

closed 2024-10-21 by sun-xiangyu