espressif/esp-dl #84: Problem with template function nn::mul2d

michalber, 2021-12-15:
When using Mul2D layer there is compile error in running [nn:mul2d function](https://github.com/espressif/esp-dl/blob/master/include/layer/dl_layer_mul2d.hpp#L143). I am using C++17, with ESP32. Problem is occuring either when creating layer with `inplace` option or without. Tried to comment [`if constexpr` statement](https://github.com/espressif/esp-dl/blob/master/include/nn/dl_nn_mul2d.hpp#L71) but it didn't help.

```cpp
no matching function for call to 'mul2d<true>(dl::Tensor<short int>&, dl::Tensor<short int>&, dl::Tensor<short int>&, const dl::Activation<short int>*&, const std::vector<int>&)'
```

I will edit this issue if I found cause of this bug.

TiramisuJ, 2021-12-16:
1. Thank you for your feedback, we will fix the bugs soon.
2. `nn::mul2d<true>(this->output_exponent, input0, input1, this->activation, assign_core);` 
this function will malloc the output tensor, and return it. so this approach is slower

TiramisuJ, 2021-12-16:
The bugs have been fixed

closed 2021-12-20 by michalber