espressif/esp-dl #85: Logistic (sigmoid) activation

dankal444, 2021-12-23:
Hi
Is it possible to have logistic (sigmoid) function activation in esp-dl? 

Can't find it in [layers](https://github.com/espressif/esp-dl/tree/master/include/layer) or [nn](https://github.com/espressif/esp-dl/tree/master/include/nn) folders. It is quite popular activation function, so I guess you must have some recipe how to implement it.

TiramisuJ, 2021-12-23:
Sure, We will add the related nonlinear functions recently

red-stream, 2022-01-10:
I want it too!  used to solve a multi-label image classification problem.

jonathanMerlingPal, 2022-09-14:
Hi All,
I need to convert a model which contains the following layers: Sigmoid, Resize, Pow in order to run it on esp32-s3-eye.
Those layers currently aren't supported on esp-dl.
How do one add new layer types so that the calibrator in the quantization tool will support the above layers? I've tried but got to this "calibrator.pyd" file which I can't edit. Any workarounds?

Thx in advance,
Jonathan

closed 2024-10-21 by sun-xiangyu