espressif/esp-dl #87: cannot build esp-dl

TecDroiD, 2022-01-12:
hi,

for some reason I cannot build any of the examples nor tutorial or anything else.. 
what am i doing wrong?

What id did so far:
tecdroid@tecdroid:~/esp$ git clone https://github.com/espressif/esp-dl.git
Klone nach 'esp-dl' ...
remote: Enumerating objects: 2129, done.
remote: Counting objects: 100% (50/50), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 2129 (delta 16), reused 50 (delta 16), pack-reused 2079
Empfange Objekte: 100% (2129/2129), 348.35 MiB | 23.79 MiB/s, fertig.
Löse Unterschiede auf: 100% (967/967), fertig.
Aktualisiere Dateien: 100% (224/224), fertig.
tecdroid@tecdroid:~/esp$ git clone https://github.com/espressif/esp-idf.git
Klone nach 'esp-idf' ...
remote: Enumerating objects: 299621, done.
remote: Counting objects: 100% (99/99), done.
remote: Compressing objects: 100% (64/64), done.
remote: Total 299621 (delta 43), reused 87 (delta 34), pack-reused 299522
Empfange Objekte: 100% (299621/299621), 169.30 MiB | 19.71 MiB/s, fertig.
Löse Unterschiede auf: 100% (221502/221502), fertig.
tecdroid@tecdroid:~/esp$ cd esp-idf
tecdroid@tecdroid:~/esp/esp-idf$ ./install.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.9.2
"python" has been detected
Installing ESP-IDF tools

.....
All done! You can now run:

  . ./export.sh

tecdroid@tecdroid:~/esp/esp-idf$ . ./export.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.9.2

....

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

tecdroid@tecdroid:~/esp/esp-idf$ cd ..
tecdroid@tecdroid:~/esp$ cd esp-dl
tecdroid@tecdroid:~/esp/esp-dl$ cd tutorial
tecdroid@tecdroid:~/esp/esp-dl/tutorial$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/tecdroid/esp/esp-dl/tutorial/build' not found. Nothing to clean.
Executing action: set-target

....

-- Components: app_trace app_update asio bootloader bootloader_support bt cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mdns mqtt newlib nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/tecdroid/esp/esp-idf/components/app_trace /home/tecdroid/esp/esp-idf/components/app_update /home/tecdroid/esp/esp-idf/components/asio /home/tecdroid/esp/esp-idf/components/bootloader /home/tecdroid/esp/esp-idf/components/bootloader_support /home/tecdroid/esp/esp-idf/components/bt /home/tecdroid/esp/esp-idf/components/cmock /home/tecdroid/esp/esp-idf/components/coap /home/tecdroid/esp/esp-idf/components/console /home/tecdroid/esp/esp-idf/components/cxx /home/tecdroid/esp/esp-idf/components/driver /home/tecdroid/esp/esp-idf/components/efuse /home/tecdroid/esp/esp-idf/components/esp-tls /home/tecdroid/esp/esp-idf/components/esp32 /home/tecdroid/esp/esp-idf/components/esp_adc_cal /home/tecdroid/esp/esp-idf/components/esp_common /home/tecdroid/esp/esp-idf/components/esp_eth /home/tecdroid/esp/esp-idf/components/esp_event /home/tecdroid/esp/esp-idf/components/esp_gdbstub /home/tecdroid/esp/esp-idf/components/esp_hid /home/tecdroid/esp/esp-idf/components/esp_http_client /home/tecdroid/esp/esp-idf/components/esp_http_server /home/tecdroid/esp/esp-idf/components/esp_https_ota /home/tecdroid/esp/esp-idf/components/esp_https_server /home/tecdroid/esp/esp-idf/components/esp_hw_support /home/tecdroid/esp/esp-idf/components/esp_lcd /home/tecdroid/esp/esp-idf/components/esp_local_ctrl /home/tecdroid/esp/esp-idf/components/esp_netif /home/tecdroid/esp/esp-idf/components/esp_phy /home/tecdroid/esp/esp-idf/components/esp_pm /home/tecdroid/esp/esp-idf/components/esp_ringbuf /home/tecdroid/esp/esp-idf/components/esp_rom /home/tecdroid/esp/esp-idf/components/esp_serial_slave_link /home/tecdroid/esp/esp-idf/components/esp_system /home/tecdroid/esp/esp-idf/components/esp_timer /home/tecdroid/esp/esp-idf/components/esp_websocket_client /home/tecdroid/esp/esp-idf/components/esp_wifi /home/tecdroid/esp/esp-idf/components/espcoredump /home/tecdroid/esp/esp-idf/components/esptool_py /home/tecdroid/esp/esp-idf/components/expat /home/tecdroid/esp/esp-idf/components/fatfs /home/tecdroid/esp/esp-idf/components/freemodbus /home/tecdroid/esp/esp-idf/components/freertos /home/tecdroid/esp/esp-idf/components/hal /home/tecdroid/esp/esp-idf/components/heap /home/tecdroid/esp/esp-idf/components/http_parser /home/tecdroid/esp/esp-idf/components/idf_test /home/tecdroid/esp/esp-idf/components/ieee802154 /home/tecdroid/esp/esp-idf/components/json /home/tecdroid/esp/esp-idf/components/log /home/tecdroid/esp/esp-idf/components/lwip /home/tecdroid/esp/esp-dl/tutorial/main /home/tecdroid/esp/esp-idf/components/mbedtls /home/tecdroid/esp/esp-idf/components/mdns /home/tecdroid/esp/esp-idf/components/mqtt /home/tecdroid/esp/esp-idf/components/newlib /home/tecdroid/esp/esp-idf/components/nvs_flash /home/tecdroid/esp/esp-idf/components/openssl /home/tecdroid/esp/esp-idf/components/openthread /home/tecdroid/esp/esp-idf/components/partition_table /home/tecdroid/esp/esp-idf/components/perfmon /home/tecdroid/esp/esp-idf/components/protobuf-c /home/tecdroid/esp/esp-idf/components/protocomm /home/tecdroid/esp/esp-idf/components/pthread /home/tecdroid/esp/esp-idf/components/sdmmc /home/tecdroid/esp/esp-idf/components/soc /home/tecdroid/esp/esp-idf/components/spi_flash /home/tecdroid/esp/esp-idf/components/spiffs /home/tecdroid/esp/esp-idf/components/tcp_transport /home/tecdroid/esp/esp-idf/components/tcpip_adapter /home/tecdroid/esp/esp-idf/components/tinyusb /home/tecdroid/esp/esp-idf/components/ulp /home/tecdroid/esp/esp-idf/components/unity /home/tecdroid/esp/esp-idf/components/usb /home/tecdroid/esp/esp-idf/components/vfs /home/tecdroid/esp/esp-idf/components/wear_levelling /home/tecdroid/esp/esp-idf/components/wifi_provisioning /home/tecdroid/esp/esp-idf/components/wpa_supplicant /home/tecdroid/esp/esp-idf/components/xtensa
-- Configuring done
CMake Error at /home/tecdroid/esp/esp-idf/tools/cmake/component.cmake:475 (add_library):
  Cannot find source file:

    /home/tecdroid/esp/esp-dl/tutorial/model/mnist_coefficient.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  main/CMakeLists.txt:19 (idf_component_register)

CMake Error at /home/tecdroid/esp/esp-idf/tools/cmake/component.cmake:475 (add_library):
  No SOURCES given to target: __idf_main
Call Stack (most recent call first):
  main/CMakeLists.txt:19 (idf_component_register)

-- Build files have been written to: /home/tecdroid/esp/esp-dl/tutorial/build
cmake failed with exit code 1

ESP-YHY, 2022-01-13:
Have you set target, like
`idf.py set-target esp32`

BTW, please use IDF release/v4.4

ESP-YHY, 2022-01-14:
> Please offer your entire log when you build. It seems lilke python environment problem.
> 
> BTW, please use IDF release/v4.4

Please do `pip install numpy`, then try again.

TecDroiD, 2022-01-14:
this works! stupid me: pip install is not pip3 install

ESP-YHY, 2022-02-16:
@TecDroiD Please close the issue if it has been solved.

closed 2022-02-25 by Auroragan

tooniesnguyen, 2025-03-22:
Sorry, I can run esp-dl with esp32 normally (not s3 or p4 ) and what do I need to change to run it. Do you guys have specific instructions?