espressif/esp-dl #94: Save enrolled face id in SD card instead of internal flash

mike-2020, 2022-04-11:
Hello,

I'd like to save enrolled face id in SD card, intead of internal flash (which has limited size). Is there a possible way to do it?

I checked class FaceRecognizer, but not found its implementation. So, it seems I cannot custermize the code by myself.

Thanks
Mike

yehangyang, 2022-05-17:
Sorry, you cannot customize the code. We'll redesign the API for this situation,

mike-2020, 2022-05-18:
Thanks. When will the new design be ready?

MichaelDu9226, 2023-12-28:
@yehangyang When will the function of saving and loading face recognition models from SD cards be completed?

Or how can I modify the version of idf4.4 to ensure that the face model saved in flash can be reloaded after restarting.
Currently, our project is based on idf4.4, but it may not be possible to migrate to idf5.0 in the short term.

Relate this issue:
https://github.com/espressif/esp-who/issues/254

sun-xiangyu, 2024-10-21:
Please try new esp-dl. Now you can save and load face id from flash or sd cards.

closed 2024-10-21 by sun-xiangyu