espressif/esp-dl #98: The example of esp-dl/tools/tensorflow_to_onnx does not support MatMul and Softmax yet

QimingLee, 2022-06-30:
We complete the  example and run the program, but it shows "Generating the quantization table:
MatMul is not supported on esp-dl yet
Softmax is not supported on esp-dl yet"

Traceback (most recent call last):
  File "D:/esp-dl-master/esp-dl-master/tools/quantization_tool/examples/example.py", line 39, in <module>
    calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)
  File "calibrator.py", line 333, in calibrator.Calibrator.generate_quantization_table
  File "calibrator.py", line 222, in calibrator.Calibrator.generate_output_model
  File "calibrator.py", line 217, in calibrator.Calibrator.check_model
ValueError: current model is not supported by esp-dl

Raven888888, 2023-04-06:
@Auroragan when can it support more layers like MatMul and Clip in calibrator? These are very common layers and the calibrator is too outdated

Farzinkh, 2023-07-08:
Hi, I was successful in implementing Convolution,MatMul and Softmax [check out here](https://github.com/Farzinkh/AI_EDGE/tree/main/ESP-Dl/Hand_gesture_recognition).

Raven888888, 2023-07-09:
@Farzinkh impressive! I sure will try out!

SamsonAdem, 2023-11-08:
Is this resolved??

AlanG2004, 2024-06-05:
> Hi, I was successful in implementing Convolution,MatMul and Softmax [check out here](https://github.com/Farzinkh/AI_EDGE/tree/main/ESP-Dl/Hand_gesture_recognition).

Where specifically should I look? The repository you provided is hard to follow, please let me know I am having the same issue and it is frustrating. Thank you.

AlanG2004, 2024-06-06:
> We complete the example and run the program, but it shows "Generating the quantization table: MatMul is not supported on esp-dl yet Softmax is not supported on esp-dl yet"
> 
> Traceback (most recent call last): File "D:/esp-dl-master/esp-dl-master/tools/quantization_tool/examples/example.py", line 39, in calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path) File "calibrator.py", line 333, in calibrator.Calibrator.generate_quantization_table File "calibrator.py", line 222, in calibrator.Calibrator.generate_output_model File "calibrator.py", line 217, in calibrator.Calibrator.check_model ValueError: current model is not supported by esp-dl

Hello were you able to solve this problem? I am running into the same issue any help would be appreciated @QimingLee @Farzinkh

sun-xiangyu, 2024-12-17:
Please try new esp-dl. Now MatMul and Softmax have been supported.

closed 2024-12-17 by sun-xiangyu