espressif/esp-dl #101: tools/quantization_tool/examples/example.py failed

lijyigithub, 2022-09-03:
```
Traceback (most recent call last):
  File "/home/li/esp32/esp-who/components/esp-dl/tools/quantization_tool/examples/example.py", line 16, in <module>
    from calibrator import *
ImportError: /home/li/esp32/esp-who/components/esp-dl/tools/quantization_tool/examples/../linux/calibrator.so: undefined symbol: _Py_CheckRecursionLimit
```

env: python 3.10

tried to install python3.7.13 but failed because of no specified version of numba

liminghui-sandra, 2022-09-05:
Hi, pls check env requirements:

> Environment Requirements:
> Python == 3.7
> [Numba](https://github.com/numba/numba) == 0.53.1
> [ONNX](https://github.com/onnx/onnx) == 1.9.0
> [ONNX Runtime](https://github.com/microsoft/onnxruntime) == 1.7.0
> [ONNX Optimizer](https://github.com/onnx/optimizer) == 0.2.6
> You can install python packages with requirement.txt:
> > pip install -r requirement.txt

eldorado2802, 2022-10-26:
Environment requirements are all satisfied, but this issue still seems to be cropping up. Trying on macOS

Traceback (most recent call last):
  File "example.py", line 16, in <module>
    from calibrator import *
ModuleNotFoundError: No module named 'calibrator'

ali-espressif, 2022-10-27:
> 

https://github.com/espressif/esp-dl/issues/93#issuecomment-1292936169

eldorado2802, 2022-10-27:
> #93 (comment)

Do I use dl-lib while cloning the repo? I'm not sure what I need to do here, I already have all the modules installed

MichelBerg, 2024-01-18:
> ```
> Traceback (most recent call last):
>   File "/home/li/esp32/esp-who/components/esp-dl/tools/quantization_tool/examples/example.py", line 16, in <module>
>     from calibrator import *
> ImportError: [...]

Do you still got that error? I tried it in google colab, to create an environment (with the requirements.txt) but somehow it isn't working. I wonder if somebody got it working…

This is my environment:
```
blinker==1.4
dbus-python==1.2.18
distro==1.7.0
jeepney==0.7.1
keyring==23.5.0
launchpadlib==1.10.16
lazr.restfulclient==0.14.4
lazr.uri==1.0.6
llvmlite==0.36.0
numba==0.53.1
numpy==1.21.6
onnx==1.9.0
onnxoptimizer==0.2.6
onnxruntime==1.7.0
protobuf==3.20.1
PyGObject==3.42.1
PyJWT==2.3.0
python-apt==2.4.0+ubuntu2
SecretStorage==3.3.1
six==1.16.0
typing_extensions==4.7.1
wadllib==1.3.6
```

lidar-guy, 2024-07-26:
also having this issue on macOS.  i've tried everything; please help

BlueSkyB, 2024-07-29:
The quantization tool for ESP-DL only supports Linux and Windows platforms now.