espressif/esp-dl #102: How to convert ONNX model to work with ESP-DL

skodippily, 2022-09-21:
Hello, 

I'm very new to machine learning. I have a working object detection model for the onnx and I tried it on the PC. May I have some instructions on how to use this to detect objects from esp32-s3. I do the quantization successfully, but when I call the _check_model_ function it has a large list of operators which is not supported on esp-dl yet.

[image]

If someone can help me with this error much appreciate with it. 

Thank you

Auroragan, 2022-09-21:
Hi, can you provide your python script?

skodippily, 2022-09-21:
> Hi, can you provide your python script?

> Hi, can you provide your python script?

I modified https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/examples/example.py file as this to test 

```
import numpy as np
import onnx
import pickle
import torch
import onnxruntime as rt

import os
import sys
import platform
system_type = platform.system()
path = f'../{system_type.lower()}'
if system_type == 'Windows':
    path = path.replace('/', '\\')
sys.path.append(path)
sys.path.append("..")

from optimizer import *
from calibrator import *
from evaluator import *

if __name__ == "__main__":

    # Optimize the onnx model
    # model_path = 'mnist_model_example.onnx'
    model_path = 'resnet50_192.onnx'
    optimized_model_path = optimize_fp_model(model_path)
    print("Optimized model path - ",optimized_model_path)

    model_proto = onnx.load(optimized_model_path)

    print('Generating the quantization table:')
    calib = Calibrator('int16', 'per-tensor', 'minmax')
    calib.check_model(model_proto)
    # calib.set_providers(['CPUExecutionProvider'])
    # calib.generate_quantization_table(model_proto, calib_dataset, pickle_file_path)

```

and https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/optimizer.py file as this

```
import numpy as np
import os
import collections

import onnx
from onnx import helper, numpy_helper, version_converter
from onnx import TensorProto
from onnxruntime.transformers.onnx_model import OnnxModel
from onnxsim import simplify

import onnxoptimizer

# check model format
def check_model_extension(model_path):
    # Split the extension from the path and normalise it to lowercase.
    ext = os.path.splitext(model_path)[-1].lower()

    # Now we can simply use != to check for inequality, no need for wildcards.
    if (ext != ".onnx"):
        raise Exception(model_path, "is an unknown file format. Use the model ending with .onnx format")

    if not os.path.exists(model_path):
[... 129 lines cut ...]
    model_proto = model_simp

    # Convert static batch to dynamic batch
    model_proto = convert_model_batch_to_dynamic(model_proto)

    # Optimizer
    model_proto = onnxoptimizer.optimize(model_proto,
                                       ['fuse_bn_into_conv', 'fuse_add_bias_into_conv', 'fuse_pad_into_conv',
                                        'fuse_matmul_add_bias_into_gemm', 'fuse_transpose_into_gemm',
                                        'eliminate_nop_flatten', 'eliminate_nop_pad', 'eliminate_nop_transpose',
                                        'eliminate_unused_initializer', 'eliminate_duplicate_initializer'])
    # Add Gemm name
    add_fused_gemm_name(model_proto)

    # Fuse bn into conv or gemm
    fuse_bn(model_proto)

    # Save optimized model
    model_name = model_path.split(".onnx")
    model_path = model_name[0] + "_simplified_optimized.onnx"
    onnx.save(model_proto, model_path)

    return model_path

```

skodippily, 2022-09-21:
I modified this https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/examples/example.py file to  test my code as in the zip file

[example_optimizer.zip](https://github.com/espressif/esp-dl/files/9614673/example_optimizer.zip)

and https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/optimizer.py also modified make the onnx model simplified
[optimizer.zip](https://github.com/espressif/esp-dl/files/9614707/optimizer.zip)

Auroragan, 2022-09-21:
you mentioned that you do the quantization successfully, did you mean running generate_quantization_table successfully or this resnet50_192.onnx model is already quantized?

skodippily, 2022-09-21:
Actually, I got one of the pre-trained models from PyTorch, and it was not quantized. then I convert to ONNX and do the quantization there. Is there any way that I can double check the model is quantized or not.

skodippily, 2022-09-21:
> generate_quantization_table

"generate_quantization_table" failed because of the missing operators.

Auroragan, 2022-09-21:
ok, then unfortunately this model cannot be deployed using esp-dl since these operations are not implemented, please check https://github.com/espressif/esp-dl/tree/master/include/layer for the supported operations.

skodippily, 2022-09-21:
oh.. thank you for your help. I was trying to create customer layers but didn't work well for me. Is there any model architecture that I need to follow if I want to detect objects for the library? like rrcnn or etc.

Auroragan, 2022-09-21:
better use a model architecture that all layers in the model are supported.

skodippily, 2022-09-21:
okay I will close the issue then. thank you