espressif/esp-dl #103: quantization tool failed at eva.generate_quantized_model

skodippily, 2022-09-28:
Hello,
Error while executing the line below in [example.py](https://github.com/espressif/esp-dl/blob/master/tools/quantization_tool/examples/example.py). 
 `eva.generate_quantized_model(model_proto, pickle_file_path)`
I slightly modify example.py to load yolov2tiny model to it.

error:
[image]

Optimized model file satisfied all the layers defined in the link [here](https://github.com/espressif/esp-dl/tree/master/include/layer)

Could you please help me with this problem.

Auroragan, 2022-10-11:
HI, what is scalarPreprocessor_scale/bias? Can you find them in the onnx graph?

skodippily, 2022-10-11:
Yes, those are under MUL and ADD layers
[image]

I'm trying to do a project base on object detection using ESP-DL. Most of the pre-trained models are not supported with built-in operators right? So the only model that I could find is this [tinyyolov2](https://github.com/onnx/models/blob/main/vision/object_detection_segmentation/tiny-yolov2/model/tinyyolov2-7.onnx) model.  If you know any other pre-trained model supported ESP-DL, please let me know.

Auroragan, 2022-10-11:
are these two nodes for normalization?

skodippily, 2022-10-11:
Not quite sure about that. I cannot find any description of each layer and its task. 
It's a pre-trained model, and haven't modified anything on it. The only thing that I noticed is when I do the optimization, BatchNormalization layer will fuse to CONV layer.

sun-xiangyu, 2024-10-21:
Please try [esp-ppq](https://github.com/espressif/esp-ppq) again.

closed 2024-10-21 by sun-xiangyu