espressif/esp-dl #107: Linking Error with Auto-load

zRedShift, 2022-11-30:
```
[7/9] Linking CXX executable FAILED:
...
/home/ronen/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(app_main.cpp.obj):(.literal._ZN2dl5layer6Conv2DIaaE4callERNS_6TensorIaEEbRKSt6vectorIiSaIiEE[_ZN2dl5layer6Conv2DIaaE4callERNS_6TensorIaEEbRKSt6vectorIiSaIiEE]+0xc): undefined reference to `_ZN2dl4tool5cache13autoload_funcEmmmm'
/home/ronen/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(app_main.cpp.obj): in function `_ZN2dl6TensorIaE14malloc_elementEb':
/home/ronen/workspace/nanit.lite.ai/components/esp-dl/include/typedef/dl_variable.hpp:368: undefined reference to `_ZN2dl4tool5cache13autoload_funcEmmmm'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

to prevent the linking error from happening I had to statically disable calls to `dl::tool::cache::autoload_func`:
https://github.com/espressif/esp-dl/compare/master...zRedShift:esp-dl:master

Also some fixes for `printf` warnings/errors

alibukharai, 2022-12-05:
@zRedShift I guess you are not using correct version of idf. Try by using idf Version 4.4

zRedShift, 2022-12-10:
@alibukharai You're right, I was using v5.1 (`esp-idf` master), any plans to move this library to support `esp-idf v5.0` now that it's been released?

alibukharai, 2022-12-10:
@zRedShift Yes, In future v5 will be avaiable for esp-dl. :)

Auroragan, 2023-02-24:
please check branch: idfv5.0

closed 2023-02-24 by Auroragan