espressif/esp-dl #112: esp32s3 recognizer undefine function

ajtwlsdlqp, 2023-02-09:
Hi,
I tried to porting esp32-s3-eye example to my target board 
target board also same esp32s3 N8R8
so, i need your help.

Eclipse (2022-09 (4.25.0)) 
ESP-IDF 4.4.3
ESP32S3-N8R8

human face detection, color detection, move detection are working well on my board,
but when i try to add recognition, i got this type error

d:/develop_tool/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: 
my_Proj_Path\libmfn.a(mfn112_1x_model_s8.cpp.obj): in function `int* std::__copy_move<false, true, std::random_access_iterator_tag>::__copy_m<int>(int const*, int const*, int*)':
/home/yuanjiong/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_algobase.h:368: undefined reference to `dl::nn::add2d(dl::Tensor<signed char>&, dl::Tensor<signed char>&, dl::Tensor<signed char>&, dl::Activation<signed char> const*, std::vector<int, std::allocator<int> > const&, int)'

there are no error when i try to include header files
[image]

my partition tble
[image]

i thinks build size also enough
[image]

i add esp-dl to submodule and i have that static library
[image]

when i call
#if CONFIG_MFN_V1
#if CONFIG_S8
    FaceRecognition112V1S8 *recognizer = new FaceRecognition112V1S8();
#elif CONFIG_S16
    FaceRecognition112V1S16 *recognizer = new FaceRecognition112V1S16();
#endif
#endif

this function that error came out..

Thanks, 
Best Regards

Auroragan, 2023-02-09:
Hi,
Can you run the origin example https://github.com/espressif/esp-dl/tree/master/examples/face_recognition successfully?

ajtwlsdlqp, 2023-02-09:
Hi, Thanks for reply
now i'm try that example to upload

[image]

[image]

[image]

there are some warning (looks purple color ) but it's working well 
my target board has no lcd to check it so i try terminal version

ajtwlsdlqp, 2023-02-16:
solved. 
not a bug

closed 2023-02-16 by ajtwlsdlqp