espressif/esp-dl #114: ESP32S3 detector_color abort() was called

ajtwlsdlqp, 2023-02-21:
Hi,
I tried to porting esp32-s3-eye example to my target board
target board also same esp32s3 N8R8
so, i need your another help.

Eclipse (2022-09 (4.25.0))
ESP-IDF 4.4.3
ESP32S3-N8R8

This issue ( https://github.com/espressif/esp-dl/issues/112#issue-1577338864) reason of duplicated cmake path 

This time issue for ColorDetector
when i use this object, esp bord called abort() function and reset

"ColorDetector detector_color;" declare this object as global_value
and make handler to call the object and to detection work it 
same as example

at that time
handler Thread's stack : 6K
before called "detector_color" object 
esp_get_free_heap_size : 6,421,507
esp_get_free_internal_heap_size : 41,516 
esp_get_minimum_free_heap_size : 6,280,163

after called "detector_color"
esp_get_free_heap_size : 6,409,763
esp_get_free_internal_heap_size : 29,768 
esp_get_minimum_free_heap_size : 6,278,095

i think heap memorry and stack are enough, but i don't know why rtos called abort()

allways core 0 called abort() function at PC 0x420E 5126

this is part of by build map file's that address area 
 .text._ZNSt7__cxx1110moneypunctIwLb0EED0Ev
                0x00000000420e50bc       0x28 d:/develop_tool/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/lib/no-rtti\libstdc++.a(monetary_members.o)
                0x00000000420e50bc                std::__cxx11::moneypunct<wchar_t, false>::~moneypunct()
 .text          0x00000000420e50e4       0x75 d:/develop_tool/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/no-rtti\libgcc.a(_fixunsdfdi.o)
                0x00000000420e50e4                __fixunsdfdi
 *fill*         0x00000000420e5159        0x3 
 .text._Z12abort_returnIiET_v
                0x00000000420e515c        0x9 esp-idf/cxx/libcxx.a(cxx_exception_stubs.cpp.obj)
 *fill*         0x00000000420e5165        0x3 
 .text.abort_expect_void
                0x00000000420e5168        0x9 esp-idf/cxx/libcxx.a(cxx_exception_stubs.cpp.obj)
                0x00000000420e5168                __wrap___register_frame_table
                0x00000000420e5168                __wrap___deregister_frame
                0x00000000420e5168                __wrap___cxa_call_unexpected
                0x00000000420e5168                abort_expect_void
                0x00000000420e5168                __wrap___register_frame
 *fill*         0x00000000420e5171        0x3 

i just code like this (same as example )
`std::vector<std::vector<color_detect_result_t>> &results = detector_color.detect((uint16_t *)frame->buf, {(int)frame->height, (int)frame->width, 3});

for (int i = 0; i < results.size(); ++i)
{
  draw_color_detection_result((uint16_t *)frame->buf, (int)frame->height, (int)frame->width, results[i], draw_lcd_colors[i % draw_colors_num]); 
}`

please, check it once
Thanks,
Best Regards

ajtwlsdlqp, 2023-02-23:
Error state update : 

abort() was called at PC 0x420e5292 on core 0
**0x420e5292: int abort_return<int>() at D:/Develop_Tool/esp-idf-v4.4.3/components/cxx/cxx_exception_stubs.cpp:33**

Backtrace: 0x40375f3a:0x3fcda870 0x4037eab1:0x3fcda890 0x40386362:0x3fcda8b0 0x420e5292:0x3fcda920 0x420e52bf:0x3fcda940 0x420caf5f:0x3fcda960 0x420cab5e:0x3fcda980 0x4202ff8e:0x3fcda9a0 0x4203002a:0x3fcda9e0 0x42030d43:0x3fcdaa00 0x42031b16:0x3fcdab10 0x4200b011:0x3fcdac00 0x420098ca:0x3fcdad00 0x40381e25:0x3fcdad20
0x40375f3a: panic_abort at D:/Develop_Tool/esp-idf-v4.4.3/components/esp_system/panic.c:402

0x4037eab1: esp_system_abort at D:/Develop_Tool/esp-idf-v4.4.3/components/esp_system/esp_system.c:128

0x40386362: abort at D:/Develop_Tool/esp-idf-v4.4.3/components/newlib/abort.c:46

0x420e5292: int abort_return<int>() at D:/Develop_Tool/esp-idf-v4.4.3/components/cxx/cxx_exception_stubs.cpp:33

0x420e52bf: __wrap__Unwind_Resume_or_Rethrow at D:/Develop_Tool/esp-idf-v4.4.3/components/cxx/cxx_exception_stubs.cpp:158

0x420caf5f: __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:90

0x420cab5e: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54

0x4202ff8e: __gnu_cxx::new_allocator<int>::allocate(unsigned int, void const*) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/ext/new_allocator.h:111
 (inlined by) std::allocator_traits<std::allocator<int> >::allocate(std::allocator<int>&, unsigned int) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/alloc_traits.h:436
 (inlined by) std::_Vector_base<int, std::allocator<int> >::_M_allocate(unsigned int) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h:296
 (inlined by) void std::vector<int, std::allocator<int> >::_M_realloc_insert<int>(__gnu_cxx::__normal_iterator<int*, std::vector<int, std::allocator<int> > >, int&&) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/vector.tcc:427

0x4203002a: void std::vector<int, std::allocator<int> >::emplace_back<int>(int&&) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/vector.tcc:109

0x42030d43: std::vector<int, std::allocator<int> >::push_back(int&&) at /home/gansichen/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/8.4.0/bits/stl_vector.h:1091
 (inlined by) ColorDetector::color_detection_forward(dl::Tensor<unsigned char>&, int) at /home/gansichen/Workspace/projects/aigit/master/dl/components/color_detect/color_detector.cpp:476

0x42031b16: ColorDetector::detect(unsigned short*, std::vector<int, std::allocator<int> >, std::vector<int, std::allocator<int> >) at /home/gansichen/Workspace/projects/aigit/master/dl/components/color_detect/color_detector.cpp:830

0x4200b011: apCameraLoop() at D:/SW_Work/my_proj/main/ap/ap_camera.cpp:334
ap_camera.cpp:334 = "std::vector<std::vector<color_detect_result_t>> &results = detector_color.detect((uint16_t *)frame->buf, {(int)frame->height, (int)frame->width, 3} );"

0x420098ca: threadCamera(void const*) at D:/SW_Work/my_proj/main/ap/ap.cpp:147 (discriminator 1)
ap.cpp:147 = "apCameraLoop();" inside this loop "detector_color" function called

0x40381e25: vPortTaskWrapper at D:/Develop_Tool/esp-idf-v4.4.3/components/freertos/port/xtensa/port.c:131

ajtwlsdlqp, 2023-02-23:
Situation Update :: 

I thinks it happend by memorry size.

[image]

change the number of detecting target color 
system reset error dosn't appred more

is there any option to change memory uses?
allocator use psram or allocator take more time but ues less memory

and can i request information of  the color threshold?
The threshold of each color is composed of 6 numbers but i don't have any idea each number's meaning 
is it R/G/B colors min/max value?

Auroragan, 2023-03-02:
Hi,
Can you provide some more details(your function and configuration) so that I can reproduce the problem?

Which allocator do you want to change?

For the color threshold, it stands for threshold in hsv colorspace
{hmin, hmax, smin, smax, vmin, vmax}
can check rgb<->hsv chapter in https://docs.opencv.org/4.x/de/d25/imgproc_color_conversions.html#color_convert_rgb_hsv
the data type is same as what is described in '8-bit image'
Maybe it's better to change this to rgb threshold for users.

ajtwlsdlqp, 2023-03-02:
Hi @Auroragan 
Thanks for help.

my main problem is when my esp32s3 board operating this code 
` 
std::vector<std::vector<color_detect_result_t>> &results = detector_color.detect((uint16_t *)frame->buf, {(int)frame->height, (int)frame->width, 3} );
`
at that time core get panniced and do the reset 
not after that code just inside that code

so i do reduce number of detector_color object's  register_color // sizeof() -> only one 
`       
while(1)
{
    err = detector_color.delete_color();
    if( err == -1)  break;  // endof color idx;
}

 //  for (int i = 0; i < detect_color_info.size(); ++i)
{ 
    detector_color.register_color( detect_color_info[info.color_result].color_thresh, detect_color_info[info.color_result].area_thresh, detect_color_info[info.color_result].name );
}

for( int i=0; i<6; i++)
{
    info.threshold_colors[i] = detect_color_info[info.color_result].color_thresh[i];
}
`
and then it happened dosn't appeard

i think it happend by memorry 
so i want know that if there are any way allocator use less memorry or can use psram's memorry

because, at the top of issue, i can check that 
"esp_get_free_heap_size()" result is enough 

Thanks,
Best Regards

closed 2024-10-21 by sun-xiangyu