espressif/esp-dl #116: Face Detection example runs but not Face Recognition

danielyang2000, 2023-03-14:
Board: ESP32-S3-WROOM Cam (only testing the non-camera stuff right now)
IDE: ESP-IDF Extension on VSCode

Hello,

I can't get the face recognition example to work on my board. After the flash stages I get the following output:

WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in idf_size.py
Total sizes:
Used static IRAM:   61742 bytes ( 300498 remain, 17.0% used)
      .text size:   60715 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   24124 bytes ( 321732 remain, 7.0% used)
      .data size:   10916 bytes
      .bss  size:   13208 bytes
Used Flash size : 2433247 bytes
      .text     :  492987 bytes
      .rodata   : 1940004 bytes
Total image size: 2505905 bytes (.bin may be padded larger)

And then the monitor stage gets stuck at the last line in the following:

(base) PS C:\Users\danie\esp-dl-master\examples\face_recognition> set IDF_PATH=C:\Users\danie\esp\esp-idf
(base) PS C:\Users\danie\esp-dl-master\examples\face_recognition> C:\Users\danie\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\danie\esp\esp-idf\tools\idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\Users\danie\esp-dl-master\examples\face_recognition\build\face_recognition.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17a4
load:0x403c9700,len:0xd68
load:0x403cc700,len:0x308c
entry 0x403c9930
I (24) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (25) boot: compile time 18:54:26
I (25) boot: chip revision: v0.1
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz     
I (37) boot.esp32s3: SPI Mode       : QIO        
I (41) boot.esp32s3: SPI Flash Size : 4MB        
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 factory          factory app      00 00 00010000 003b6000
I (70) boot:  1 nvs              WiFi data        01 02 003c6000 00004000
I (77) boot:  2 fr               Unknown data     01 06 003ca000 00020000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=1abach (109484) map
I (114) esp_image: segment 1: paddr=0002abd4 vaddr=3fc92300 size=02aa4h ( 10916) load
I (116) esp_image: segment 2: paddr=0002d680 vaddr=40374000 size=02998h ( 10648) load
I (122) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=439f8h (276984) map
I (170) esp_image: segment 4: paddr=00073a20 vaddr=40376998 size=0b884h ( 47236) load
I (185) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (186) octal_psram: vendor id    : 0x0d (AP)
I (191) octal_psram: dev id       : 0x02 (generation 3)
I (197) octal_psram: density      : 0x03 (64 Mbit)
I (202) octal_psram: good-die     : 0x01 (Pass)
I (207) octal_psram: Latency      : 0x01 (Fixed)
I (213) octal_psram: VCC          : 0x01 (3V)
I (218) octal_psram: SRF          : 0x01 (Fast Refresh)
I (224) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (229) octal_psram: BurstLen     : 0x01 (32 Byte)
I (235) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (241) octal_psram: DriveStrength: 0x00 (1/1)
I (247) esp_psram: Found 8MB PSRAM device
I (251) esp_psram: Speed: 80MHz
I (255) cpu_start: Pro cpu up.
I (258) cpu_start: Starting app cpu, entry point is 0x403753d0
0x403753d0: call_start_cpu1 at C:/Users/danie/esp/esp-idf/components/esp_system/port/cpu_start.c:142

The monitor will output "...APP CPU up" after "call_start_cpu1" line for the face detection example. The face recognition example never outputs that and is struck on "...cpu_start.c:142" line. 

Initially, I thought it was the issue with "WARNING: Given section not found in any memory region." from ESP-IDF size. However, I tried commenting out most of the main_app.c in face recognition example to the point where that warning doesn't show up anymore. However, monitor state still gets stuck on "...cpu_start.c" line. 

Any ideas or suggestions on how to fix both issues: memory region warning and app not starting correctly (stuck on cup_start.c:142 line)? Thank you in advance.

danielyang2000, 2023-03-14:
ESP-IDF Version 5.0.1

Auroragan, 2023-03-15:
Hi, are you using branch idfv5.0?

danielyang2000, 2023-03-15:
Yes, I tried both the master branch and idfv5.0 branch.

Auroragan, 2023-03-23:
Hi, for master branch, need to use idf release/v4.4,
For idfv5.0 branch, could you try on idf branch: release/v5.0? It works fine for me on this branch.

danielyang2000, 2023-03-24:
Thank you for the help. I still don't know what's causing this issue but I was able to run the face recognition code in the face-detection example folder. It must've been the configuration or something else set up in the face-recognition example folder that's breaking the code.

Mr-Techtron, 2023-08-31:
@Auroragan 
For the master branch in esp-who, I am using the release/v4.4 and it is able to detect face, nose, and eyes as expected using the esp-who framework, but when I switch to idfv5.0 branch in the esp-who and switch to release/v5.0 then it fails to detect face, nose and eyes using the webUI.

sun-xiangyu, 2024-10-21:
please try new esp-dl

closed 2024-10-21 by sun-xiangyu