espressif/esp-dl #117: Sending ESP32-Cam FrameBuffer into face detection example's infer() function

danielyang2000, 2023-03-15:
board: ESP32-S3-WROOM Cam board
IDE: ESP-IDF extention on VSCode 
version: 5.0.1

Hello,
I plan to modify the face detection example to infer on the images captured from the board OV2640 camera. How would I do this?

This is what I see on lines 25 and 26
std::list<dl::detect::result_t> &candidates = s1.infer((uint8_t *)IMAGE_ELEMENT, {IMAGE_HEIGHT, IMAGE_WIDTH, IMAGE_CHANNEL});
std::list<dl::detect::result_t> &results = s2.infer((uint8_t *)IMAGE_ELEMENT, {IMAGE_HEIGHT, IMAGE_WIDTH, IMAGE_CHANNEL}, candidates);

Would I change them to:

// assuming camera initialization is successful 
fb = esp_camera_fb_get();
std::list<dl::detect::result_t> &candidates = s1.infer((uint8_t *)fb->buf, {(int) fb->height, (int) fb->width, 3});
std::list<dl::detect::result_t> &results = s2.infer((uint8_t *)fb->buf, {(int) fb->height, (int) fb->width, 3}, candidates);

basically changing IMAGE_HEIGHT to fb->height, IMAGE_WIDTH to fb->width, IMAGE_Element to fb->buf, and IMAGE_CHANNEL to 3? 
I tried this, and this runs but doesn't seem to predict the face when I put my face in front of the camera. Am I missing anything?

Thank you

Auroragan, 2023-03-15:
HI, you can check examples with camera input here:
https://github.com/espressif/esp-who/tree/master/examples/human_face_detection

danielyang2000, 2023-03-16:
Problem resolved. Turns out the face model only takes rgb888 images. I had to use fmt2rgb888 function to convert OV2640 frame buffer to rgb888 format

closed 2023-03-23 by Auroragan